adafruit/Adafruit_Wippersnapper_Arduino #432: Pico W Infinite Reboot Loop After Wippersnapper Install

indecorous, 2023-05-17:
**Describe the bug**
Hello all,
I hope this finds you well.

I install the firmware, .json file, then restart. The Pico enters an endless reboot loop. I checked out the .json file. The username and IO I.D. are correct.  There are no error messages to note. I’ve followed the directions and have attempted multiple times. I’ve even used a few different Pico Ws that are known to be completely functional. 

Any help would be greatly appreciated.
Thanks for your time.

LLAP

***Arduino board***
Rapberry Pi Pico W

**To Reproduce**
Steps to reproduce the behavior:
I followed the instructions as per the Adafruit site.

**Expected behavior**
I would it expect it to connect to Adafruit IO.

**Which components are connected to your device**
Currently, nothing is connected to the Pico W other than the USB cable to interface.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11 Pro 64-bit (22H2), Raspberry Pi Bullseye, Linux Mint
 - Browser: Google Chrome and Chromium
 - Version: 113.0.5672.93 (Official Build) (64-bit)

**Smartphone (please complete the following information):**
-NA

**Additional context**
None noted.

brentru, 2023-05-30:
@indecorous Looking into this today, I found that on my network the **third** try passes through to Adafruit IO. 

It should attempt to retry the connection to Adafruit IO 5 times prior to rebooting the watchdog timer, but it seems this does not occur:

```
12:35:09.274 -> Attempting to connect to WiFi...
12:35:09.274 -> Attempting to connect to WiFi...
12:35:17.587 -> Attempting to connect to WiFi...
12:35:17.587 -> Attempting to connect to Adafruit IO...
WDT EXPIRED - REBOOT
12:35:37.315 -> Attempting to connect to WiFi...
12:35:37.315 -> Attempting to connect to WiFi...
12:35:45.623 -> Attempting to connect to WiFi...
12:35:45.623 -> Attempting to connect to Adafruit IO...
WDT EXPIRED - REBOOT
12:36:05.389 -> Attempting to connect to WiFi...
12:36:05.389 -> Attempting to connect to WiFi...
12:36:13.702 -> Attempting to connect to WiFi...
12:36:13.702 -> Attempting to connect to Adafruit IO...
12:36:19.669 -> Registering hardware with WipperSnapper...
12:36:19.669 -> Registering hardware with IO...
```

On a board with INVALID ADAFRUIT IO CREDENTIALS, I got the reboot loop you are discussing:

```
12:51:44.655 -> Attempting to connect to WiFi...
12:51:44.655 -> Attempting to connect to Adafruit IO...
12:51:45.269 -> Port number: 8883
12:51:45.269 -> CID: io-wipper-rpi-pico-w7117500
12:51:45.269 -> Connecting to: io.adafruit.com
12:51:49.333 -> Connect result: 1
12:51:49.333 -> MQTT connect packet:
DATA HERE SNIPPED FOR GITHUB
12:51:49.370 -> Client sendPacket returned: 84
12:51:49.370 -> Read data:		  [0x20], 
12:51:49.370 -> Packet Type:		  [0x20], 
12:51:49.370 -> Read data:		  [0x02], 
12:51:49.370 -> Packet Length:	2
12:51:49.370 -> Read data:		  [0x00],   [0x05], 
12:51:49.370 -> Unable to connect to Adafruit IO MQTT, retrying in 3 seconds...
12:51:52.372 -> Port number: 8883
12:51:52.372 -> CID: io-wipper-rpi-pico-w7117500
12:51:52.372 -> Connecting to: io.adafruit.com
REBOOT
12:51:56.240 -> Adafruit.io WipperSnapper
```

@indecorous  Please check your Adafruit IO Username and Key along with your WiFi SSID and Password please. Also, try another wifi network.

brentru, 2023-05-30:
Note: I've split the bug I noticed off into https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/issues/435

indecorous, 2023-05-30:
Interesting. Thanks for investigating.
I double checked the secrets.json. The login credentials for IO and WiFi seem to be correct. I've followed the instructions meticulously. 

I attempted this from a different WiFi network but got the same results. I can tell you that no other devices have an issue connecting to my router. Not sure where to go from here.

I've noticed that user accounts can get hosed in Terminal Services and need to be recreated from scratch. Is it possible something like that has happened here: a corrupted user account? Sorry if that's a dumb question. I'm not sure how the back-end of IO works.

brentru, 2023-06-01:
> I double checked the secrets.json. The login credentials for IO and WiFi are correct. I've followed the instructions meticulously.

Bizzare, and you're unable to use either the Arduino Serial Monitor or https://www.serialterminal.com/ to get debug output?

> If it is truly unrelated to the bug report, I have a question. I've noticed that user accounts can get hosed in Terminal Services and need to be recreated from scratch. I realize this isn't Terminal Services or domain controller related, but is it possible something like that has happened here: a corrupted user account? Sorry if that's a dumb question. I'm not sure how the back-end of IO works.

If you fill out a query on http://io.adafruit.com/support and link this bug report, I can ask the web team to take a look into your account.

indecorous, 2023-06-03:
Another potentially silly question: I’ve noticed that the usernames in the forum tend to start with a lower-case letter. Is it possible that the capital letter in my username is causing issues? 

Also ediazcoon, in the Adafruit forum, explained that they started having the same issue with their Pico after that last update of Wippsnapper. It seems worth mentioning.

https://forums.adafruit.com/viewtopic.php?p=974533#p974533

brentru, 2023-06-07:
> Another potentially silly question: I’ve noticed that the usernames in the forum tend to start with a lower-case letter. Since IO and the Adafruit user account use the same login credentials, is it possible that the capital letter in my username is causing problems?

I asked our development team, the username's capitalization should not matter here.

@indecorous  I have a request that'll help me replicate this.

Either on support (I just sent you a new message so you can attach it) or here, could you attach the CURRENT.UF2 file that's present on your Pico W? 

To get this file:
* Start with your Pico W unplugged from USB. Hold down the BOOTSEL button, and while continuing to hold it (don't let go!), plug the Pico W into USB. Continue to hold the BOOTSEL button until the RPI-RP2 drive appears!
* Copy the CURRENT.UF2 file from RPI-RPI2 to your desktop.
* Attach it here or in an email

indecorous, 2023-06-07:
[INFO_UF2.TXT](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/files/11680370/INFO_UF2.TXT)

I don't see one called CURRENT.UF2. The above INFO_UF2,TXT is it. There's also an INDEX.HTM that takes you to https://www.raspberrypi.com/documentation/microcontrollers/?version=E0C9125B0D9B . 

I flashed wippersnapper.rpi_pico_w.1.0.0-beta.63.uf2 .

magher, 2023-06-08:
I have the same problem. Here are some screen shots from what the USB drive contains after flashing with Wippersnapper and result in Thonny after the 'secrets' file was replaced with the one generated on the Adafruit site.
[image]

[image]

magher, 2023-06-08:
I've used the same secrets file with micro python on the same device (reflashed of course) and it works just fine.

brentru, 2023-06-08:
@magher Thanks for the serial monitor output, I'm starting to feel like this issue is linked to https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/issues/435 and will look into resolving that first to see if it helps.

brentru, 2023-06-09:
@magher  Both of these issues seem different. It is difficult to debug two people with different (but "similar") issues in the same thread.

Could you please open a new issue on https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/issues/new?assignees=&labels=&projects=&template=bug_report.md&title= so I can help you there and @indecorous  in here?

Thanks!

brentru, 2023-06-09:
@indecorous  Can we try this before I implement a new "safe mode":

* Load the "Pico W Nuke UF2" so we can rebuild with a fresh filesystem: https://learn.adafruit.com/getting-started-with-raspberry-pi-pico-circuitpython/circuitpython#flash-resetting-uf2-3083182
* Load the WipperSnapper UF2
* Load your secrets.json file
* Please attach a copy of `wipper_boot_out.txt` from the WIPPER drive

indecorous, 2023-06-09:
I’ve rebooted it but there isn’t a whippersnapper_boot_out.txt this time. I have attached a couple of screenshots.

Also, I’m getting a permissions pop up every time I load a file to the rp2040 Wippersnapper. Not sure if this is normal or not. Included in attached screen shots.

[image]
[image]

brentru, 2023-06-09:
@indecorous  I see `wipper_boot_out.txt` on the file explorer screenshot.

I'm not sure about the permissions popup as of right now, I haven't experienced that before.

indecorous, 2023-06-09:
LOL... sorry about that. It wasn't there the first couple of reboots so I missed it on the last screen shot. My bad. I promise I'm not huffing glue...

[wipper_boot_out.txt](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/files/11709776/wipper_boot_out.txt)

indecorous, 2023-06-09:
Just tried in Linux Mint and Raspberry Pi OS. I got the same results. I've added the wipper_boot_out.txt from the Linux Mint attempt.
[wipper_boot_out.txt](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/files/11710011/wipper_boot_out.txt)

brentru, 2023-06-12:
@indecorous  Since I can't replicate this, I'm going to look at implementing two ideas:

- If the serial is connected, I am going to try adding a feature to disable the watchdog timer ("the reboot") unless a user physically presses the REBOOT button. 
  - This will allow the serial terminal to stay open long enough to get a debug log that's useful for both of us
- I want to persist boot_out.txt across boots, it currently gets deleted and re-written on each application boot. If there was a error with the previous boot, I want to print that out to the .txt file and persist it to the next boot.

indecorous, 2023-06-12:
Very cool. Just let me know how to proceed.

As always, thanks.

indecorous, 2023-06-20:
Hello,
Just curios if there has been any progress or if I should be doing anything to help. I want to make sure you're not waiting on me to do something.

Let me know when you get a chance.

Thanks again.

indecorous, 2023-06-28:
What's up with the crickets?... 
I'm still here...

brentru, 2023-06-30:
I am the sole developer of this firmware. I was focused on releasing beta 64, released this week, which kicks off display integration so we can do some awesome visualization: https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/releases/tag/1.0.0-beta.64

With the display feature shipped, I am starting to loop back to unresolved and difficult-to-reproduce issues like this one after the USA Holiday this upcoming week.

indecorous, 2023-07-07:
I'm sorry Adafruit  is so understaffed. Have you asked for help?

brentru, 2023-12-06:
Addressed in #514 , please try the latest version of Adafruit WipperSnapper

closed 2023-12-06 by brentru