adafruit/Adafruit_Wippersnapper_Arduino #434: Compilation error

rocketrob, 2023-05-27:
Compilation error: no matching function for call to 'Wippersnapper_ESP8266::Wippersnapper_ESP8266(const char [10], const char [33], const char [11], const char [11])'

***Arduino board***
Amica NodeMCU Generic ESP8266

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
Compile and upload WipperSnapper to device for adding to My Device page on WipperSnapper

**Which components are connected to your device**
Which components are set up on the WipperSnapper device page? n/a

**Screenshots** 
If applicable, add screenshots to help explain your problem. n/a

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]  - Windows 11
 - Browser [e.g. chrome, safari] Chrome
 - Version [e.g. 22]

**Smartphone (please complete the following information):** n/a
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

brentru, 2023-05-30:
@rocketrob  I am going to reproduce this but I need more information. On this menu, which board did you select:

[image]

rocketrob, 2023-05-31:
I tried both the generic ESP8266 and the adafruit feather hazzah ESP8266

How that helps

Rob

brentru, 2023-06-01:
@rocketrob You're attempting to load WipperSnapper on an Amica NodeMCU Generic ESP8266, right? What board do you typically select for this? 

btw - this board is unsupported, I assume you are adding it to WipperSnapper and following the guide: https://learn.adafruit.com/how-to-add-a-new-board-to-wippersnapper

brentru, 2023-07-03:
@rocketrob  I do not get a compilation error when I compile for the Huzzah ESP8266. 

When I attempt to compile for the "Generic ESP8266" board, I get a compilation error. This error is expected for not-yet-supported boards such as the Generic ESP8266 platform. 

```
/Users/brentrubell/Documents/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/Wippersnapper.cpp: In member function 'bool Wippersnapper::generateDeviceUID()':
/Users/brentrubell/Documents/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/Wippersnapper.cpp:1684:17: error: 'BOARD_ID' was not declared in this scope
 1684 |   WS._boardId = BOARD_ID;
      |                 ^~~~~~~~
In file included from /Users/brentrubell/Documents/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/Wippersnapper.cpp:34:
/Users/brentrubell/Documents/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/Wippersnapper.cpp: In function 'void printDeviceInfo()':
/Users/brentrubell/Documents/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/Wippersnapper.cpp:2490:20: error: 'BOARD_ID' was not declared in this scope
 2490 |   WS_DEBUG_PRINTLN(BOARD_ID);
      |                    ^~~~~~~~
```

To resolve it, closely follow the steps in the Adafruit Learning System Guide for "Modifying Firmware | How to Add a Board to WipperSnapper": https://learn.adafruit.com/how-to-add-a-new-board-to-wippersnapper/modifying-firmware

If you need assistance porting a board to WipperSnapper, please open a new issue in this repository.

closed 2023-07-03 by brentru