adafruit/Adafruit_Wippersnapper_Arduino #436: Adafruit I/O does not connect to Raspberry Pi Pico W board

magher, 2023-06-10:
**Describe the bug**
After following the on-line guide at Adafruit to connect the Pico to Adafruit I/O and rebooting the device the output in the serial monitor of Thonny is 'Attempting to connect to Wifi and ends with an error (see pic)

***Arduino board***
Raspberry Pi PicoW

**To Reproduce**
Steps to reproduce the behavior:
1. Go to (https://io.adafruit.com/[account]/devices/new)
2. Follow the instructions for setting up a new device
3. Replace the secrets file with the one generated from step 6 
4. When done (step 8) reboot the board

**Expected behavior**
I expect it to work of course :) 
Note: 
- Secrets file also tested with same device and micro python and the credentials is verified to work. 
- When rebooting the board in step 8, no pop-up shows to be able to save the device
- The USB drive is called 'USB Drive' not 'Wipper'

**Which components are connected to your device**
None

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows11 Home, version 10.0.22621 build 22621
 - Browser: Chrome Version 114.0.5735.110 (Official Build) (64-bit)

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.
Serial monitor output...
[image]

Contents on the USB device...
[image]

brentru, 2023-06-12:
@magher Are you able to try a different WiFi network or a mobile WiFi network hotspot?

magher, 2023-06-13:
Tried it on my iPhone network and it seems to work better.... Can it be a firewall issue?

Registering hardware with WipperSnapper...
Registering hardware with IO...
Encoding registration request...Encoding registration msg...Published!
Polling for registration message response...2
GOT Registration Response Message:
Hardware Response Msg:
	GPIO Pins: 31
	Analog Pins: 4
	Reference voltage: 3.30v
Completed registration process, configuration next!
Polling for message containing hardware configuration...
Polling for message containing hardware configuration...
cbSignalTopic: New Msg on Signal Topic
2 bytes.
decodeSignalMsg
cbSignalMsg
Sub-messages found: 1
Signal Msg Tag: Pin Configuration
Initial Pin Configuration Complete!
Publishing to pin config complete...
Hardware configured successfully!
Registration and configuration complete!
Running application...
PING!
PING!
PING!

magher, 2023-06-13:
Hmmm, I was sure that it was 2.4GHz but apparently it is 5GHz

magher, 2023-06-13:
It is an ASUS meshed router with 3 nodes if it makes any difference

magher, 2023-06-15:
Progress ??

magher, 2023-06-16:
Switched network to 2.4GHz but same result

brentru, 2023-06-16:
@magher  I wonder if the Pico W's board support package for Arduino that we use (https://github.com/earlephilhower/arduino-pico) has issues connecting to your mesh wifi setup. I don't have a mesh setup so it's not possible for me to replicate this issue.

Are you familiar with how to use Arduino IDE to upload a sketch? We could try uploading a simple WiFi connection sketch to the Pico W to isolate if its the WiFi, or if it's WipperSnapper.

magher, 2023-06-16:
Ok. Do you have a sketch in mind or should I whip something up?
//M

brentru, 2023-06-16:
@magher I'd suggest:

1) Install Arduino Pico: https://arduino-pico.readthedocs.io/en/latest/install.html
2) In Arduino IDE, navigate to File->Examples->WiFi->WiFiClient
3) Modify the sketch to use your SSID/password and upload it.
4) See if it connects to your home network

magher, 2023-06-16:
Running 5 GHz
Unable to connect to network, rebooting in 10 seconds...
Unable to connect to network, rebooting in 10 seconds...
Unable to connect to network, rebooting in 10 seconds...
.
.
.

Running 2.4 GHz
IP address: 
192.168.50.152
connecting to djxmmx.net:17
sending data to server
receiving from remote server
"It's not about where your dreams take you, but where you take your dreams..."
	 - Anonymous
closing connection
etc....
.
.
.

magher, 2023-06-16:
Works on 2.4GHz. Updated the ticket.
//M

I’ll do that
//M

brentru, 2023-06-20:
@magher  So, 2.4GHz appears to work for the Pico W and 5GHz does not. This is expected behavior as the chipset does not support 5GHz.

Is there anything else to resolve before I close out this issue?

zardoz99, 2023-07-21:
I can confirm that the same problem occurs here, using wippersnapper.rpi_pico_w.1.0.0-beta.66.uf2

 I also have an Asus mesh network and I have tried it on the main wireless LAN and also a guest wireless LAN. These are 2.4 GHz networks. 

I have tried using two different Pico W boards, with exactly the same results.

The symptoms are generally the same as the initial report but VERY occasionally it does manage to connect to the Wi-Fi and start to initiate the dialogue with adafruit.io. It then drops out again and goes back into the failure loop .

I have tried using my Android phone as a hotspot but no success there either.

After a number of iterations of the loop, the internal FAT file system ends up getting corrupted and I lose the secrets.json file. I then have to restart all over again from scratch.

zardoz99, 2023-07-21:
Out of sheer stubbornness and determination, I once again tried using my phone as a hotspot and changed the secrets.json file appropriately. For reasons unknown, this time the Pico W was actually able to connect to adafruit.io using the hotspot Wi-Fi. I'm really not sure what this tells us. Clearly the software is incompatible with an Asus mesh network and/or router(s). This is unfortunately a bit of a showstopper for me.

zardoz99, 2023-07-21:
Having made a single successful connection I have since updated to the latest available beta. Reverting back to the Wi-Fi network provided by the Asus mesh results in the original problem reappearing. Here is what I get on the serial connection over USB.

Adafruit.io WipperSnapper
-------Device Information-------
Firmware Version: 1.0.0-beta.67
Board ID: rpi-pico-w
Adafruit.io User: zardoz99
WiFi Network: Shadow-2G
MAC Address: 28:CD:C1:0B:79:68
-------------------------------
Generating device's MQTT topics...
Running Network FSM...
Connecting to WiFi...
Attempting to connect to WiFi...
Attempting to connect to WiFi...
Attempting to connect to WiFi...
Attempting to connect to WiFi...
Attempting to connect to WiFi...
ERROR: Unable to connect to WiFi!
ERROR [WDT RESET]: ERROR: Unable to connect to WiFi, rebooting soon...

brentru, 2023-07-21:
@zardoz99  Which Asus router do you have?

zardoz99, 2023-07-21:
```
AiMesh router
--
RT-AX86U
Current Version : 3.0.0.4.388_23285-g5068da5

AiMesh Node
RT-AX86S
Current Version : 3.0.0.4.388_23285-g5068da5
RT-AC86U 
Current Version : 3.0.0.4.386_51915-g43629d2
```

zardoz99, 2023-07-21:
In case it is helpful, I am including a segment of log from the router which shows that the Pico W does in fact connect to the network and gets offered at DHCP response providing it with the correct IP address.

```
Jul 21 18:46:10 RT-AX86U-C280-818D6EF-C/192.168.0.1 kernel: wl0: random key value: 31392003804AAE4137920216CB4E42CAF948C257E64B0A5FA23150D793D91EE7
Jul 21 18:46:10 RT-AX86U-C280-818D6EF-C/192.168.0.1 wlceventd: wlceventd_proc_event(503): eth6: Deauth_ind 28:CD:C1:0B:79:68, status: 0, reason: Unspecified reason (1), rssi:0
Jul 21 18:46:10 RT-AX86U-C280-818D6EF-C/192.168.0.1 wlceventd: wlceventd_proc_event(539): eth6: Auth 28:CD:C1:0B:79:68, status: Successful (0), rssi:0
Jul 21 18:46:10 RT-AX86U-C280-818D6EF-C/192.168.0.1 wlceventd: wlceventd_proc_event(568): eth6: Assoc 28:CD:C1:0B:79:68, status: Successful (0), rssi:-56
Jul 21 18:46:10 RT-AX86U-C280-818D6EF-C/192.168.0.1 hostapd: eth6: STA 28:cd:c1:0b:79:68 IEEE 802.11: associated
Jul 21 18:46:10 RT-AX86U-C280-818D6EF-C/192.168.0.1 hostapd: eth6: STA 28:cd:c1:0b:79:68 RADIUS: starting accounting session 1F4DD85A11B4AABE
Jul 21 18:46:10 RT-AX86U-C280-818D6EF-C/192.168.0.1 hostapd: eth6: STA 28:cd:c1:0b:79:68 WPA: pairwise key handshake completed (RSN)
Jul 21 18:46:13 RT-AX86U-C280-818D6EF-C/192.168.0.1 dnsmasq-dhcp[4211]: DHCPDISCOVER(br0) 28:cd:c1:0b:79:68
Jul 21 18:46:13 RT-AX86U-C280-818D6EF-C/192.168.0.1 dnsmasq-dhcp[4211]: DHCPOFFER(br0) 192.168.0.100 28:cd:c1:0b:79:68
Jul 21 18:46:29 RT-AX86U-C280-818D6EF-C/192.168.0.1 kernel: wl0: random key value: 3DA087F613E5BD4D6DA80099F58C9FEAD0F58E2DA85E0357B4FAE658A1AD6868
Jul 21 18:46:29 RT-AX86U-C280-818D6EF-C/192.168.0.1 wlceventd: wlceventd_proc_event(503): eth6: Deauth_ind 28:CD:C1:0B:79:68, status: 0, reason: Unspecified reason (1), rssi:-57
Jul 21 18:46:29 RT-AX86U-C280-818D6EF-C/192.168.0.1 wlceventd: wlceventd_proc_event(539): eth6: Auth 28:CD:C1:0B:79:68, status: Successful (0), rssi:-57
Jul 21 18:46:29 RT-AX86U-C280-818D6EF-C/192.168.0.1 wlceventd: wlceventd_proc_event(568): eth6: Assoc 28:CD:C1:0B:79:68, status: Successful (0), rssi:-57
Jul 21 18:46:29 RT-AX86U-C280-818D6EF-C/192.168.0.1 hostapd: eth6: STA 28:cd:c1:0b:79:68 IEEE 802.11: associated
Jul 21 18:46:29 RT-AX86U-C280-818D6EF-C/192.168.0.1 hostapd: eth6: STA 28:cd:c1:0b:79:68 RADIUS: starting accounting session 1F4DD85A11B4AABE
Jul 21 18:46:29 RT-AX86U-C280-818D6EF-C/192.168.0.1 hostapd: eth6: STA 28:cd:c1:0b:79:68 WPA: pairwise key handshake completed (RSN)
Jul 21 18:46:32 RT-AX86U-C280-818D6EF-C/192.168.0.1 dnsmasq-dhcp[4211]: DHCPDISCOVER(br0) 28:cd:c1:0b:79:68
Jul 21 18:46:32 RT-AX86U-C280-818D6EF-C/192.168.0.1 dnsmasq-dhcp[4211]: DHCPOFFER(br0) 192.168.0.100 28:cd:c1:0b:79:68
Jul 21 18:46:37 RT-AX86U-C280-818D6EF-C/192.168.0.1 kernel: wl0: random key value: 902F8A4102A8F4DE5F3001C31A3CB4A6733C8775C26B0313095217B077CDFC67
Jul 21 18:46:37 RT-AX86U-C280-818D6EF-C/192.168.0.1 wlceventd: wlceventd_proc_event(503): eth6: Deauth_ind 28:CD:C1:0B:79:68, status: 0, reason: Unspecified reason (1), rssi:0
Jul 21 18:46:37 RT-AX86U-C280-818D6EF-C/192.168.0.1 wlceventd: wlceventd_proc_event(539): eth6: Auth 28:CD:C1:0B:79:68, status: Successful (0), rssi:0
Jul 21 18:46:37 RT-AX86U-C280-818D6EF-C/192.168.0.1 wlceventd: wlceventd_proc_event(568): eth6: Assoc 28:CD:C1:0B:79:68, status: Successful (0), rssi:-56
Jul 21 18:46:37 RT-AX86U-C280-818D6EF-C/192.168.0.1 hostapd: eth6: STA 28:cd:c1:0b:79:68 IEEE 802.11: associated
Jul 21 18:46:37 RT-AX86U-C280-818D6EF-C/192.168.0.1 hostapd: eth6: STA 28:cd:c1:0b:79:68 RADIUS: starting accounting session 1F4DD85A11B4AABE
Jul 21 18:46:37 RT-AX86U-C280-818D6EF-C/192.168.0.1 hostapd: eth6: STA 28:cd:c1:0b:79:68 WPA: pairwise key handshake completed (RSN)
Jul 21 18:46:40 RT-AX86U-C280-818D6EF-C/192.168.0.1 dnsmasq-dhcp[4211]: DHCPDISCOVER(br0) 28:cd:c1:0b:79:68
Jul 21 18:46:40 RT-AX86U-C280-818D6EF-C/192.168.0.1 dnsmasq-dhcp[4211]: DHCPOFFER(br0) 192.168.0.100 28:cd:c1:0b:79:68
Jul 21 18:46:40 RT-AX86U-C280-818D6EF-C/192.168.0.1 kernel: wl0: random key value: 534D21B18577FBC80EDF025017F14E33A04AFC11A1C80555B5F294181C73853C
Jul 21 18:46:40 RT-AX86U-C280-818D6EF-C/192.168.0.1 wlceventd: wlceventd_proc_event(520): eth6: Disassoc 28:CD:C1:0B:79:68, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Jul 21 18:46:40 RT-AX86U-C280-818D6EF-C/192.168.0.1 wlceventd: wlceventd_proc_event(520): eth6: Disassoc 28:CD:C1:0B:79:68, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8), rssi:0
Jul 21 18:46:40 RT-AX86U-C280-818D6EF-C/192.168.0.1 hostapd: eth6: STA 28:cd:c1:0b:79:68 IEEE 802.11: disassociated
Jul 21 18:46:40 RT-AX86U-C280-818D6EF-C/192.168.0.1 hostapd: eth6: STA 28:cd:c1:0b:79:68 IEEE 802.11: disassociated
Jul 21 18:46:45 RT-AX86U-C280-818D6EF-C/192.168.0.1 wlceventd: wlceventd_proc_event(539): eth6: Auth 28:CD:C1:0B:79:68, status: Successful (0), rssi:0
Jul 21 18:46:45 RT-AX86U-C280-818D6EF-C/192.168.0.1 hostapd: eth6: STA 28:cd:c1:0b:79:68 IEEE 802.11: associated
Jul 21 18:46:45 RT-AX86U-C280-818D6EF-C/192.168.0.1 wlceventd: wlceventd_proc_event(568): eth6: Assoc 28:CD:C1:0B:79:68, status: Successful (0), rssi:-55
Jul 21 18:46:45 RT-AX86U-C280-818D6EF-C/192.168.0.1 hostapd: eth6: STA 28:cd:c1:0b:79:68 RADIUS: starting accounting session 9EC7216FDFA81397
Jul 21 18:46:45 RT-AX86U-C280-818D6EF-C/192.168.0.1 hostapd: eth6: STA 28:cd:c1:0b:79:68 WPA: pairwise key handshake completed (RSN)
[... cut ...]
Jul 21 18:49:27 RT-AX86U-C280-818D6EF-C/192.168.0.1 hostapd: eth7: STA 1e:d5:2b:b2:81:c8 WPA: pairwise key handshake completed (RSN)
Jul 21 18:49:27 RT-AX86U-C280-818D6EF-C/192.168.0.1 dnsmasq-dhcp[4211]: DHCPREQUEST(br0) 192.168.0.148 1e:d5:2b:b2:81:c8
Jul 21 18:49:27 RT-AX86U-C280-818D6EF-C/192.168.0.1 dnsmasq-dhcp[4211]: DHCPACK(br0) 192.168.0.148 1e:d5:2b:b2:81:c8 SM-R910
```

One of the things that I did note is that it takes 3 seconds after the connection is established for the DHCP offer to go out. Is this too long? I have no mechanism to change it, if it is.

brentru, 2023-07-24:
> From what I am seeing, I deduce that we are not waiting long enough for the network to fully establish before giving up. On the few occasions that the DHCP response arrives within 2 seconds, we do establish the connection but when it takes 3 seconds or longer we fail. Obviously I don't know how WipperSnapper is handling the connections and what timers are configured. Maybe that is something that you can experiment with.

This much can be adjusted, we can change the maximum WiFi connection attempts here: https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/blob/main/src/Wippersnapper.cpp#L2195

electron271, 2023-08-16:
I am also having this issue with a xFi router, using a hotspot also worked.

tyeth, 2023-09-21:
Just to add, the changeover for CI builds has happened, along with BSP upgrades and there has been no change in the connectivity ability. (My home 2.4GHz network broadcast from Virgin Media Hub5 is unconnectable with v72 of wippersnapper)

brentru, 2023-12-06:
@magher @zardoz99  Hi - we have just released WipperSnapper beta 74. This includes PR #514 which addresses issues with the Pico W infinite reboot loop issue caused by WiFi failing to connect.

Please give this a try and let me know if you are still experiencing issues.

closed 2023-12-06 by brentru

magher, 2023-12-07:
Hi again,

Unfortunately I still have the same problem with
wippersnapper.picow_rp2040_tinyusb.1.0.0-alpha.74.uf2

Thonny gives me...

Attempting to connect to WiFi...
Attempting to connect to WiFi...
Attempting to connect to WiFi...
ERROR: Unable to connect to WiFi!
ERROR [WDT RESET]: ERROR: Unable to connect to WiFi, rebooting soon...

It is just longer time between the reboots

//Magnus

tyeth, 2023-12-07:
Hi @magher there was a problem with 74, so please try v75 and report back. Also is that the 5GHz or 2.4GHz WiFi network?

magher, 2023-12-07:
It was the 2.4 GHz. Does it work on 5GHz?
//Magnus

tyeth, 2023-12-07:
I haven't actually tried 5ghz as the problem for me was on 2.4GHz. Try v75 and see 
https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/releases