adafruit/Adafruit_Wippersnapper_Arduino #452: I2C(and normal components) Packet Decode Errors

tyeth, 2023-07-19:
So this ticket is to cover current and future i2c packet decode errors like #408 and #437 (also showed up in #451)
I initially had a false assumption that it related to unpublished components, but as we've seen in the past it mostly related to the adafruit mqtt library and the packet size and subscription maxlength.
I think because I built at least a few times during testing on the Arduino IDE it would have used the adafruit version of the mqtt library, whereas the platformio (VSCode IDE) and github CI have both been changed to use a forked version that @brentru maintains hotfixes/WipperSnapper experiments on.
This could have led to my decode error as there are noticable changes to those maxlengths e.g. https://github.com/adafruit/Adafruit_MQTT_Library/compare/master...brentru:Adafruit_MQTT_Library:master#diff-240d52c08217f01fd327187461355707b15e7e24e0e6e3da37f0a67fa12006ceL110-L127

The Decode Error occurs at boot, after connecting to adafruit IO as part of receiving the I2C settings. This can occur when deleting a component (reset), or if wifi dropped / power loss etc, and the device will be fully functional beforehand and dead after this ocurrance. It can be fixed by removing the i2c components (to reduce the max packet size related to i2c components), and if the user is unaware then they can add all the components back again and data will flow for many more components than can be supported until when the device resets and receives the i2c settings again. This probably happens with non-i2c components too.

tyeth, 2023-08-07:
`PUBLISHING -> I2C Device Sensor Event Message...ERROR: Failed to encode and publish I2CDeviceEvent!`
Noted while near the throttle limit, that new components aren't always initialised, despite the device appearing to wait for the throttle before attempting to send i2c device message. Must be removed and re-added. This was a second BMP388 being added, but after selecting the metrics frequency and choosing add component the device was throttled and failed to acknowledge the addition successfully. It would be nice if it retried.
[image]

brentru, 2023-08-31:
This should be resolved by https://github.com/adafruit/Adafruit_MQTT_Library/pull/227 and will be included in WipperSnapper Beta 71 (released later today).

Please try beta 71 and let me know in this issue if the packet decode error has been resolved or is still persistent.

closed 2023-08-31 by brentru