adafruit/Adafruit_Wippersnapper_Arduino #456: Unable to modify MagTag file system contents

brentru, 2023-07-26:
***Arduino board***
MagTag

**To Reproduce**
Steps to reproduce the behavior:

1. Choose any of the following actions
2. rename `secrets.json` to `secrets.old`
3. edit `secrets.json` (using notepad, exiting notepad) to have API key, username, SSID, and password
4. copy (from host) new `secrets.json` over existing file
5. Attempt safe removal: right-click the safe removal icon in taskbar, and choose to eject the USB device
6. Wait 30+ seconds
7. Either press reset button on MagTag, or slide power switch to off position for at least 3 seconds

**Expected behavior**
The file volume appears, and reflects the action selected above that modified secrets.json.

**Actual Results**
The first boot flashes neopixels purple for <1 second.
No serial port is enumerated, and no mass storage appears.
At this point, if use reset button again (or power switch), WIPPER volume will reappear, but with original, unmodified contents.

Also noteworthy: The "Safe Removal" attempt never completes. It's almost as though the device is failing all write commands, and thus rejecting the QUERY_REMOVAL until the cache is flushed ... which never happens.

**Additional context**

Adafruit.io WipperSnapper
Firmware Version: 1.0.0-beta.67
Board ID: magtag
MAC Address: 7C:DF:A1:17:8E:28

tinyuf2-adafruit_magtag_29gray-0.16.0
wippersnapper.magtag.1.0.0-beta.67.uf2

brentru, 2023-07-26:
Waiting for Forum poster to provide the OS they are running.

brentru, 2023-07-31:
Firmware Version: 1.0.0-beta.67 from "wippersnapper.qtpy_esp32s2.1.0.0-beta.67.uf2"
TinyUF2 Bootloader f12dd0f - tinyusb (0.10.1-109-g9b3ec69b)
Uploaded with Windows 10

xsk8rat, 2024-02-24:
This behavior persists with

- wippersnapper.magtag.1.0.0-beta.76.uf2

- tinyuf2-adafruit_magtag_29gray-0.18.1

brentru, 2024-02-26:
@tyeth Are you able to repro this?

tyeth, 2024-02-26:
No I haven't got a MagTag, but I can look at the m4 airlift and try to reproduce on that or the pyportals later this week.

closed 2026-09-24 by brentru