adafruit/Adafruit_Wippersnapper_Arduino #465: ESP32-S2 with TFT Feather connection issue to Adafruit IO

robpSF, 2023-08-15:
**Describe the bug**
The board connects to wifi - I can see it's connected to the network - but then it'll connect momentarily to Adafruit IO and stop. 
The board then auto restarts

***Arduino board***
ESP32-S2 with TFT Feather 

**To Reproduce**
1. Go to Adafruit.io
2. add new device
3. provision the device as directed by following the steps given
4. Yeah! everything is working. 
5. Test by adding a button or LED on the dashboard
6. Yeah! It's still working - I can trigger events 
7. Unplug the USB from the device. Obviously the device goes offline because there;s no power
8. Plug in the USB.... the device powers up
9. You can see the device connects to wifi but can't connect to Adafruit.io
10. Device gives up and restarts trying to reconnect to Adafruit.io

**Expected behavior**
I expect device to come back online

**Which components are connected to your device**
It doesn't matter. I've tried with one button and then several. One LED and then several

**Screenshots**
None necessary

**Desktop (please complete the following information):**
Windows 10 - all up to date 
Chrome - all up to date

**Smartphone (please complete the following information):**
N/A

**Additional context**
This is a brand new device straight from shop. 
All works during the first install but after power is removed and then repowered it doesn't come back online again.

robpSF, 2023-08-21:
Hi has there been any progress?

robpSF, 2023-09-06:
I was hoping there might be some progress on this? Is there any update please?

tyeth, 2023-09-06:
Hi @robpSF you'll be relieved to hear there has! This last week the bug has finally been crushed.
A new release is being cut today, look out for the blog post in the next twelve hours, and that will be version 1.0.0-beta72. You can also use the Watch option on this repository to be informed of new releases etc.

robpSF, 2023-09-06:
Superb! Thank you very much!!

brentru, 2023-09-06:
@robpSF No problem! Please let us know if this is working for you.

brentru, 2023-11-02:
Closed as this has been fixed in beta 72 and resolved.

closed 2023-11-02 by brentru