adafruit/Adafruit_Wippersnapper_Arduino #467: No PlatformIO.ini entry for Pico W

bscherer-sdvi, 2023-08-16:
**Describe the bug**
Cant build for Pico W in PlatformIO (while Pico is a supported board for WipperSnapper, theres no PlatformIO.ini entry)

***Arduino board***
Pico W

**To Reproduce**
Steps to reproduce the behavior:
Trying building for Pico W in PlatformIO

**Expected behavior**
The ability to build for Pico W

**Which components are connected to your device**
Which components are set up on the WipperSnapper device page?

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

bscherer-sdvi, 2023-08-16:
I tried writing my own section but went down a dependency hell-hole where I couldnt seem to find some dependencies

```
[env:raspberrypi_pico_w]
platform = raspberrypi
board = pico
lib_ldf_mode = deep+
build_flags = -DUSE_TINYUSB=1
	-DARDUINO_RASPBERRY_PI_PICO_W
```

brentru, 2023-08-16:
@bscherer-sdvi Unfortunately,  RaspberryPi Trading Ltd. had a disagreement with PlatformIO Labs when it came to adding the Earlephilhower Arduino Pico core, the full details are on this PR: https://github.com/platformio/platform-raspberrypi/pull/36. The result is that [there is no platformio package for the rp2040 anymore. ](https://github.com/earlephilhower/arduino-pico/issues/1617#issuecomment-1657116853).

As a result, I recommend compiling WipperSnapper for PicoW using the Earlephilhower and Arduino IDE.

I can leave this issue open if you're interested in trying out max's solution in (https://github.com/earlephilhower/arduino-pico/issues/1617), otherwise I will close it.

bscherer-sdvi, 2023-08-16:
Thanks for the quick reply! I will try whats in that repo. Go ahead an close :)

bscherer-sdvi, 2023-08-16:
err I can i see

closed 2023-08-16 by bscherer-sdvi