adafruit/Adafruit_Wippersnapper_Arduino #489: Small wifi reliability enhancements

brentru, 2023-09-13:
After talking with @tyeth, we've identified two changes to make to increase wifi reliability:

* For PicoW, there's an issue [where it fails to connect to WiFi](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/issues/436#issuecomment-1648284464)
  * Within [this sample sketch](https://github.com/earlephilhower/arduino-pico/issues/1639#issuecomment-1672293436), the `WiFi` object's [setTimeout function](https://github.com/earlephilhower/arduino-pico/blob/master/libraries/WiFi/src/WiFiClass.h#L384) is called with a timeout of 20 seconds. 
  * We should try this and implement it if it fails less often
* To help find mobile hotspots or other mobile APs, we could increase the number of times WipperSnapper attempts to scan for the wireless network described in `secrets.json`.

brentru, 2023-11-02:
@tyeth  Since you can replicate the PicoW bug, do you want to take this issue on (at least the first bullet)?

brentru, 2024-08-09:
@tyeth  Do we still want to take a look at this, or close it out?

tyeth, 2024-08-09:
I still have problems connecting pico-w's, but my local network setup is currently in flux, transitioning to a mesh which brings its own fun/problems.

Think it's worth leaving open, and as you previously said, for me to test with a 20second timeout.