adafruit/Adafruit_Wippersnapper_Arduino #495: "Dimmable LED" Component on HUZZAH ESP8266 is inverted

brentru, 2023-09-27:
**Describe the bug**
The dimmable LED component's functionality is inverted when used with the Feather Huzzah ESP8266:

[image]

This may be caused due to the fact that the Adafruit Feather Huzzah ESP8266 has inverted wiring on the RED LED.

***Arduino board***
INSERT ARDUINO BOARD NAME/TYPE HERE

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new Dimmable LED Component
2. Drag the slider to 100%
3. Drag the slider back to 0%

**Which components are connected to your device**
Dimmable LED (D0)

brentru, 2023-11-02:
@tyeth  Wanna take a look at this issue?

tyeth, 2023-11-03:
This sounds like something for the analog output basics guide, so we show a normal one and mention / explain how that would play out if you had it wired oppositely (i.e. the LED would be on with the Low part of the duty cycle), like you find on some boards.

closed 2023-11-24 by brentru