adafruit/Adafruit_Wippersnapper_Arduino #500: Error with wifi connection, Metro ESP32-S2 immediately disconnects

wollstars, 2023-10-05:
**Describe the bug**
Hi, I bought an ESP32-S2 last week and have been trying to get it to accept wifi commands this week.

I first started with aREST, and noticed that it would be able to accept things like pin changes over wifi but was unable to run functions. Thinking it was potentially a library issue, I moved to Adafruit IO to see if it was a board issue or not.

I found the same issue - I could accept smaller commands over the feed, but not through the menu. Turning the LED on and off would immediately terminate the connection and my board would disconnect from USB if I had it plugged in - I think it's resetting every time.

[image]

I've tried four different power sources (3.7V 1200mAH battery, 9v 1A wall charger, 3xAA pack, over USB) and have flashed this board five times with different versions of the firmware trying to get it to work, but I keep running into the same issue regarding wifi (regardless of the network, which is always 2.4GHz) no matter what I do.

After all this testing, I'm worried it might be a hardware issue with my board. Any help would be appreciated!

***Arduino board***
Metro ESP32-S2

**To Reproduce**
Uncertain on how to reproduce. I took my device out of the box and it had this issue.

**Expected behavior**
Communication over wifi.

**Which components are connected to your device**
No components are connected to my device.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser Chrome
 - Version Most recent

**Smartphone (please complete the following information):**
 - Device: Z Flip3
 - OS: Android
 - Browser Chrome
 - Version Most recent

**Additional context**
I can't execute commands on Arduino over wifi on my device without it dropping out as well, so I think it might just be a hardware issue.

wollstars, 2023-10-06:
> @wollstars Are you able to connect this board, without any components added to it, to WipperSnapper? Does the issue only occur when you add a new component?
> 
> @lorennorman I think this device is missing a group key, the feed looks like username/feeds/ws-001 but lacks the group (device) identifier. Do you know why this may happen?

I can connect the board to Wippersnapper fine. It's just executing a command that causes it to disconnect.

brentru, 2023-10-06:
@wollstars  If you can, can you try connecting your board to a serial monitor like the Arduino IDE Serial Monitor or to https://serial.huhn.me/ (11500 baud rate). You should be able to see the debug output from the board. If you see debug output such as ping messages, try toggling the LED switch. 

Copy and paste all the error/debug output here, so we can analyze it.

wollstars, 2023-10-06:
@brentru Registering hardware with WipperSnapper...

Registering hardware with IO...

Encoding registration request...Encoding registration msg...Published!

Polling for registration message response...2

GOT Registration Response Message:

Hardware Response Msg:

	GPIO Pins: 19

	Analog Pins: 17

	Reference voltage: 2.60v

Completed registration process, configuration next!

Polling for message containing hardware configuration...

Polling for message containing hardware configuration...

Polling for message containing hardware configuration...

cbSignalTopic: New Msg on Signal Topic

27 bytes.

decodeSignalMsg

cbSignalMsg

Sub-messages found: 1

Signal Msg Tag: Pin Configuration

cbDecodePinConfigMsg

Configured digital output pin on D42

cbDecodePinConfigMsg

Configured digital output pin on D5

Initial Pin Configuration Complete!

Publishing to pin config complete...

Hardware configured successfully!

Registration and configuration complete!

Running application...

PING!

cbSignalTopic: New Msg on Signal Topic

12 bytes.

decodeSignalMsg

cbSignalMsg

Sub-messages found: 1

Signal Msg Tag: Pin Event

cbDecodeDigitalPinWriteMsg

Digital Pin Event: Set 42 to 0

PING!

PING!

PING!

IO Ban Error: "unexpected error saving data on bankersgravity/feeds/ws-001, WipperSnapper Device not found with group_key "

brentru, 2023-10-11:
@lorennorman  This looks like an issue with IO not assigning a group key? 

> IO Ban Error: "unexpected error saving data on bankersgravity/feeds/ws-002, WipperSnapper Device not found with group_key "

lorennorman, 2023-10-12:
I've *just* figured out what's going on here, sorry it took me so long to circle back!

Working on a fix now, sit tight @wollstars and I'll ping you again when it's in place.

wollstars, 2023-10-12:
@lorennorman It worked, thanks so much - I had a bit of an issue with the disconnect when I was first enabling it, so I was a bit worried, but it seems to accept commands now!

What was the issue? I'm going to try running Arduino wifi code on the board at some point, so I'd love to know what was causing the crazy disconnects in the first place.

closed 2023-10-12 by wollstars

lorennorman, 2023-10-13:
@wollstars Yessss! Grats on getting it working!

The bugs I fixed were server and web-side (that is, not device code.) I won't give too much detail, but behind the scenes we were handling "default" groups incorrectly. As a result, your WipperSnapper device inadvertently got flagged as your default group, making it impossible to delete it. Another issue is that the device settings window was ignoring server errors, it should've been showing them to you (I'll fix that soon.)

As for your hardware integrity, I *think* the device code automatically reboots itself if it sees an MQTT error (is that right, @brentru?) That means the device is fine... or at least, those disconnects were intentional and not based on faulty hardware. Of course I can't guarantee you don't **also** have faulty hardware! 😛 

❤️ We are **extremely grateful** that you patiently reported this! It helped me discover a few dozen users in the same state. I closed up the hole and fixed all the accounts yesterday, huge win. ❤️

brentru, 2023-10-13:
> As for your hardware integrity, I think the device code automatically reboots itself if it sees an MQTT error (is that right, @brentru?) That means the device is fine... or at least, those disconnects were intentional

Correct!