adafruit/Adafruit_Wippersnapper_Arduino #503: Neopixels

gomermil, 2023-10-28:
**Describe the bug**
I cannot use more then 60 pixels when I am trying to use my neopixels from my device. 

***Arduino board***
Raspberry Pi Pico W

**To Reproduce**
Steps to reproduce the behavior:
1. Go to devices
2. Click on raspberry pi pico w
3. Scroll down to living room neopixel
4. Set to more then 60 neopixels
5. Turn pixels on and nothing happens 

**Expected behavior**
The neopixels should turn on for the amount I have told it I have attached. 

**Which components are connected to your device**
Neopixels 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Microsoft Edge
 - Version: 117.0.2045.60

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

brentru, 2023-10-30:
@gomermil Hi - I am going to attempt to replicate this bug but need two details first:

Which product are you using that has 60 or more NeoPixels connected? Is it something like the [Adafruit NeoPixel Digital RGBW LED Strip - White PCB 144 LED/m - 1m](https://www.adafruit.com/product/2847)?

What version of WipperSnapper are you running on the Pico W?

gomermil, 2023-10-30:
I figured it out. I was not restarting the pico after changing the amount of pixels. Thank you for your support. VRMark Roemer

brentru, 2023-10-31:
I'm going to keep this issue open as the device should update, regardless of the amounts of pixels, and not require a reboot

brentru, 2024-08-09:
@tyeth  Want to take a crack at reproducing this issue?

brentru, 2025-03-03:
Unable to reproduce this issue, closing...

closed 2025-03-03 by brentru