adafruit/Adafruit_Wippersnapper_Arduino #522: Display WiFi RSSI level on status bar

brentru, 2023-12-11:
Add a timer callback to check the RSSI from the network adapter and display a wifi icon on the status bar.

T-Mosher, 2024-05-24:
Bumping this, it would be really nice to know the RSSI for Wippersnapper WiFi connections. Maybe make it a Component that can be used as a feed.

tyeth, 2024-05-29:
Linking https://github.com/adafruit/Wippersnapper_Components/issues/126 

 And https://github.com/tyeth/Adafruit_Wippersnapper_Arduino/tree/add-rssi

tyeth, 2024-05-31:
Just to update that this RSSI displayed in the status bar is being considered as part of the UI redesign, but won't be any time soon.
For now there is a change to print the RSSI value on each ping (MQTT keep alive every 5seconds) to the serial port as part of normal / debug logging.

tyeth, 2024-07-29:
Further note that RSSI is printed in serial monitor, both on connect and during MQTT pings. It is also displayed for each network from a wifi scan if desired SSID is not found. (Also supported with multi-wifi networks as part of #376)