adafruit/Adafruit_Wippersnapper_Arduino #527: Compilation error: exit status 1 Arduino UNO R4 WiFi

Carda1989, 2024-01-06:
**Describe the bug**
error while verify

***Arduino board***
Arduino UNO R4 WiFi

xxx are for privacy, datas are correct in the sketch

**Sketch** 
// Adafruit IO WipperSnapper Beta
//
//
// NOTE: This software is a BETA release and in active development.
// Please report bugs or errors to
// https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/issues
//
// This sketch is for devices which lack USB-MSD or LittleFS support such
// as the Arduino MKR WiFi 1010, Arduino Nano 33 IoT.
//
// Adafruit invests time and resources providing this open source code.
// Please support Adafruit and open source hardware by purchasing
// products from Adafruit!
//
// Brent Rubell for Adafruit Industries, 2021
//
// All text above must be included in any redistribution.

#include "Wippersnapper_Networking.h"

/************************ Adafruit IO Config *******************************/

// Visit io.adafruit.com if you need to create an account,
// or if you need your Adafruit IO key.
#define IO_USERNAME  "xxx"
#define IO_KEY       "xxx"

/**************************** WiFi Config ***********************************/
#define WIFI_SSID "xxx"
#define WIFI_PASS "xxx"

#include "Wippersnapper_Networking.h"
Wippersnapper_WiFi wipper(IO_USERNAME, IO_KEY, WIFI_SSID, WIFI_PASS);

void setup() {
  // Provisioning must occur prior to serial init.
  wipper.provision();

  Serial.begin(115200);
  // while (!Serial) delay(10);

  wipper.connect();
}

void loop() { wipper.run(); }

**Error**
In file included from /Users/xxx/libraries/Adafruit_WipperSnapper/src/Wippersnapper.h:35:0,
                 from /Users/xxx/libraries/Adafruit_WipperSnapper/src/Wippersnapper.cpp:34:
/Users/xxx/libraries/Adafruit_SleepyDog_Library/Adafruit_SleepyDog.h:41:2: error: #error Unsupported platform for the Adafruit Watchdog library!
 #error Unsupported platform for the Adafruit Watchdog library!
  ^~~~~
In file included from /Users/xxx/libraries/Adafruit_WipperSnapper/src/Wippersnapper.h:59:0,
                 from /Users/xxx/libraries/Adafruit_WipperSnapper/src/Wippersnapper.cpp:34:
/Users/xxx/libraries/Adafruit_WipperSnapper/src/components/ds18x20/ws_ds18x20.h:22:10: fatal error: DallasTemperature.h: No such file or directory
 #include <DallasTemperature.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1

Thanks for help

tyeth, 2024-01-08:
Hi @Carda1989 and thanks for having a go at building the WipperSnapper firmware on a new platform!
I've raised an issue for the Adafruit SleepyDog library to add support for the R4 Uno, and the other dependency is Dallas Temperature Library (we install that as part of a Continuous Integration step when doing our builds, or manually on developer machines - it's also listed in our PlatformIO dependencies as https://github.com/milesburton/Arduino-Temperature-Control-Library which depends on https://github.com/PaulStoffregen/OneWire/ ).
[image]

tyeth, 2024-01-09:
Note to future self / others: Brent mentioned that tinyUSB possibly doesn't currently support the R4, and this morning I just found this which claims to be "A tinyusb based project implementing the bootloader for Uno R4 and Portenta C33"
https://github.com/arduino/arduino-renesas-bootloader

Now checking the tinyUSB support table it does list the RA4M1 so that repos probably unnecessary

closed 2024-08-09 by brentru