adafruit/Adafruit_Wippersnapper_Arduino #542: VCNL4020 causes reboot on initialisation when only device externally plugged in (StemmaQT), works with additional StemmaQT devices on I2C Bus

tyeth, 2024-02-02:
Testing #537 on a funhouse, found it rebooted at initialisation of the VNCL4020 (I had already initialised all magic config components including an onboard i2c temp sensor).

Tested again on V76 release and same thing. Tested again with Feather S3 tft on both firmwares same issue. Instead attaching my other vncl4020 which was joined to a string (not string but star/parallel 4port i2c) of other sensors, it then worked. Testing with a single additional device before or after the vncl4020 allowed the vcnl4020 to initialise. Without it appears to be unhappy. Comparing to the vcnl4040 (near identical code/device) which works there appear to be no voltage translation transistors (chip marked SS containing two BSS138's according to schematic)

VCNL4040:
https://cdn-learn.adafruit.com/assets/assets/000/078/737/original/adafruit_products_schematic.png?1564449985
[image]

VNCL4020:
https://cdn-learn.adafruit.com/assets/assets/000/124/960/original/adafruit_products_schem.png?1696620643
[image]

tyeth, 2024-02-02:
Tagging @ladyada and @brentru

tyeth, 2024-02-02:
Not so sure... Seems like the 4020 should accept 5v on i2c
[image]

ladyada, 2024-02-02:
try to turn on SERIAL_DEBUG in the BusIO I2CDevice.c file, and look at the output :)

tyeth, 2024-02-07:
Decoded Backtrace on feather reverse-tft s3 during boot loop:
```
0x40377aae: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:471
0x4037de4d: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:84
0x40383dc5: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:81
0x40378953: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:387 (discriminator 1)
0x40383e05: free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/heap.c:39
0x420b20b5: operator delete(void*) at /builds/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++/del_op.cc:49
0x420b1495: operator delete(void*, unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++/del_ops.cc:33
0x42022983: Adafruit_VCNL4020::begin(TwoWire*, unsigned char) at /home/runner/Arduino/libraries/Adafruit_VCNL4020_Library/Adafruit_VCNL4020.cpp:64 (discriminator 1)
0x4200ab86: WipperSnapper_I2C_Driver_VCNL4020::begin() at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/components/i2c/drivers/WipperSnapper_I2C_Driver_VCNL4020.h:59
 (inlined by) WipperSnapper_Component_I2C::initI2CDevice(_wippersnapper_i2c_v1_I2CDeviceInitRequest*) at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/components/i2c/WipperSnapper_I2C.cpp:491
0x42005c72: cbDecodeI2CDeviceInitRequestList(pb_istream_s*, pb_field_iter_s const*, void**) at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/Wippersnapper.cpp:637
0x420ca843: pb_default_field_callback at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/nanopb/pb_common.c:310
0x4200d1fd: decode_callback_field at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/nanopb/pb_decode.c:767
 (inlined by) decode_field at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/nanopb/pb_decode.c:818
0x4200d5c9: pb_decode_inner at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/nanopb/pb_decode.c:1109
0x4200d622: pb_decode at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/nanopb/pb_decode.c:1185
0x42005dc8: cbDecodeSignalRequestI2C(pb_istream_s*, pb_field_iter_s const*, void**) at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/Wippersnapper.cpp:738
0x4200cff3: pb_dec_submessage at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/nanopb/pb_decode.c:1607
 (inlined by) decode_basic_field at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/nanopb/pb_decode.c:453
0x4200d10d: decode_static_field at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/nanopb/pb_decode.c:549
 (inlined by) decode_field at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/nanopb/pb_decode.c:812
0x4200d5c9: pb_decode_inner at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/nanopb/pb_decode.c:1109
0x4200d622: pb_decode at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/nanopb/pb_decode.c:1185
0x42003e66: cbSignalI2CReq(char*, unsigned short) at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/Wippersnapper.cpp:894
0x4201137d: Adafruit_MQTT::processSubscriptionPacket(Adafruit_MQTT_Subscribe*) at /home/runner/Arduino/libraries/Adafruit_MQTT_Library/Adafruit_MQTT.cpp:236
0x42011532: Adafruit_MQTT::processPacketsUntil(unsigned char*, unsigned char, unsigned short) at /home/runner/Arduino/libraries/Adafruit_MQTT_Library/Adafruit_MQTT.cpp:268
0x42011821: Adafruit_MQTT::publish(char const*, unsigned char*, unsigned short, unsigned char, bool) at /home/runner/Arduino/libraries/Adafruit_MQTT_Library/Adafruit_MQTT.cpp:382
0x420056d1: Wippersnapper::publish(char const*, unsigned char*, unsigned short, unsigned char) at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/Wippersnapper.cpp:2607
0x420058d2: Wippersnapper::publishPinConfigComplete() at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/Wippersnapper.cpp:2809
0x42006343: Wippersnapper::connect() at /home/runner/Arduino/libraries/Adafruit_Wippersnapper_Arduino/src/Wippersnapper.cpp:2771
0x42003bf2: setup() at /home/runner/work/Adafruit_Wippersnapper_Arduino/Adafruit_Wippersnapper_Arduino/examples/wippersnapper_debug/wippersnapper_debug.ino:19
0x420359a3: loopTask(void*) at /home/runner/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/cores/esp32/main.cpp:59
```

Serial Output from WipperSnapper:
```
---- Opened the serial port COM42 ----
12:33:44:675 -> SSID found!
12:33:45:682 -> Connecting to WiFi (attempt #0)
12:33:51:291 -> Connected to WiFi!
12:33:51:292 -> Connecting to AIO MQTT (attempt #0)
12:33:51:292 -> WiFi Status: 20
12:33:55:374 -> Registering hardware with WipperSnapper...
12:33:55:374 -> Registering hardware with IO...
12:33:55:374 -> Encoding registration request...Encoding registration msg...Published!
12:33:55:577 -> Polling for registration message response...2
12:33:56:577 -> GOT Registration Response Message:
12:33:56:578 -> Hardware Response Msg:
12:33:56:578 ->         GPIO Pins: 13
12:33:56:578 ->         Analog Pins: 6
12:33:56:578 ->         Reference voltage: 2.60v
[... cut ...]
12:33:56:747 -> 38 bytes.
12:33:56:747 -> cbDecodeSignalRequestI2C
12:33:56:747 -> I2C Device LIST Init Request Found!
12:33:56:747 -> EXEC: cbDecodeI2CDeviceInitRequestList
12:33:56:747 -> EXEC: New I2C Port 
12:33:56:747 ->         Port #: 0
12:33:56:747 ->         SDA Pin: 3
12:33:56:747 ->         SCL Pin: 4
12:33:56:747 ->         Frequency (Hz): 100000
12:33:56:998 -> Attempting to initialize I2C device: vcnl4020
---- Closed serial port COM42 due to disconnection from the machine ----
```

tyeth, 2024-02-07:
@brentru having looked at the two drivers (vcnl4020 vs 4040) we don't delete the i2c device in the 4040, instead just newing it up again (which seems to work reliably). 
https://github.com/adafruit/Adafruit_VCNL4020/blob/main/Adafruit_VCNL4020.cpp#L63-L66
https://github.com/adafruit/Adafruit_VCNL4040/blob/master/Adafruit_VCNL4040.cpp#L54

Not sure that should crash it, but there we go. Probably worth reporting on esp repo. Can you see if you get similar?

ladyada, 2024-02-07:
fixed
https://github.com/adafruit/Adafruit_VCNL4020/commit/aad19441e0da1f04494e6adb05d72ced0646539f
please pull and retest

tyeth, 2024-02-07:
Looks great, retested, haven't seen an error since (plenty of reboots attempted). Thanks @ladyada

closed 2024-02-07 by tyeth