adafruit/Adafruit_Wippersnapper_Arduino #545: RED LIGHT: Improve Error Messages in early failure situations and possibly compile time:

tyeth, 2024-02-08:
My fault no doubt as playing with different framework versions in platformio, but got a malloc error assigning device id, probably setup without use_psram, but all I saw as a user was "nothing on serial" (takes too long to reconnect and not repeated message) and red light. There is a mounted drive, showing valid secrets etc and bootlog looked okay (showing mac etc).

Eventually I noticed the red light stopped and started again (when rebooting), so I removed enough usb devices and shut enough things that my serial monitor was able to reconnect fast enough to catch the boot error message (I've not seen it again in ten reboots - too slow to reconnect serial monitor in windows):
```
---- Opened the serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
Adafruit.io WipperSnapper
-------Device Information-------
Firmware Version: 1.0.0-beta.76
Board ID: feather-esp32s3-tft
Adafruit.io User: tyeth
WiFi Network: THERMAL
MAC Address: F4:12:FA:59:C1:B8
-------------------------------
ESP32 CPU0 RESET REASON: NO_MEAN
ESP32 CPU1 RESET REASON: POWERON_RESET
E ERROR: Unable to create device uid, Malloc failure
ERROR [WDT RESET]: Unable to generate Device UID
---- Closed serial port COM45 due to disconnection from the machine ----
```

Solution is to at least get all messages to repeat, or provide some compile time warning if wrong setting for board (psram etc)

tyeth, 2024-02-08:
To catch the message, because at boot the device dettaches serial/msc then reattaches the serial monitor catches the reboot attach of serial then dettach and reattach again, but it takes time (like 1sec) to attach. The only way to catch the message is to start serial monitoring manually after the reboot brings up serial the first time (before dettach and reattach) then it consumes the 1second while the device quickly does the reattach, meaning it's ready to receive the early boot message.

```
---- Opened the serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
Adafruit.io WipperSnapper
-------Device Information-------
Firmware Version: 1.0.0-beta.76
Board ID: feather-esp32s3-tft
Adafruit.io User: tyeth
WiFi Network: THERMAL
MAC Address: F4:12:FA:59:C1:B8
-------------------------------
ESP32 CPU0 RESET REASON: NO_MEAN
ESP32 CPU1 RESET REASON: POWERON_RESET
E ERROR: Unable to create device uid, Malloc failure
ERROR [WDT RESET]: Unable to generate Device UID
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed the serial port COM45 ----
---- Opened the serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed the serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
**** Failed to open the serial port COM45 ****
---- Reopened serial port COM45 ----
Adafruit.io WipperSnapper
-------Device Information-------
Firmware Version: 1.0.0-beta.76
Board ID: feather-esp32s3-tft
Adafruit.io User: tyeth
WiFi Network: free4all_2G
MAC Address: F4:12:FA:59:C1:B8
-------------------------------
ESP32 CPU0 RESET REASON: NO_MEAN
ESP32 CPU1 RESET REASON: POWERON_RESET
E (4097) task_wdt: delete_entry(232): task not found
ERROR: Unable to create device uid, Malloc failure
ERROR [WDT RESET]: Unable to generate Device UID
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
---- Closed serial port COM45 due to disconnection from the machine ----
---- Reopened serial port COM45 ----
```

brentru, 2024-02-08:
@tyeth  Which board were you using? Could you share the Platformio .ini config here?

Regarding this error:
> E (4097) task_wdt: delete_entry(232): task not found

Could you split this out into another issue and label it "bug" and set status to "needs investigation" so we can investigate it more?

tyeth, 2024-02-09:
PlatformIO.ini (was swapping about platform + platform_packages and also build flags for both boards:
**s3 tft feather** and **s3 reverse tft feather**
```ini
; PlatformIO Project Configuration File for Adafruit.IO WipperSnapper Firmware
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

; PlatformIO Defaults
[platformio]
description = "Adafruit.IO WipperSnapper Firmware"
default_envs = adafruit_feather_esp32s3_tft, adafruit_magtag29_esp32s2, adafruit_feather_esp32s3, featheresp32s2, adafruit_feather_esp32s2_reversetft, adafruit_metro_esp32s2, adafruit_qtpy_esp32s2, adafruit_feather_esp32s3_reversetft, adafruit_feather_esp32s3_nopsram, adafruit_feather_esp32s2_tft, adafruit_qtpy_esp32, adafruit_qtpy_esp32c3, featheresp32, adafruit_qtpy_esp32s3_nopsram, featheresp32v2, huzzah, adafruit_pyportal_m4, adafruit_pyportal_m4_titano

; Common Build Environment
; https://docs.platformio.org/en/latest/projectconf/section_env.html
[env]
framework = arduino
monitor_speed = 115200
lib_compat_mode = strict
lib_deps =
    adafruit/Adafruit Zero DMA Library
    adafruit/Adafruit TinyUSB Library
    adafruit/Adafruit NeoPixel
    adafruit/Adafruit SPIFlash
    adafruit/Adafruit DotStar
    adafruit/ENS160 - Adafruit Fork
    adafruit/Adafruit SleepyDog Library
    adafruit/Adafruit AHTX0
    adafruit/Adafruit BME280 Library
    adafruit/Adafruit BMP280 Library
    adafruit/Adafruit BMP3XX Library
    adafruit/Adafruit DPS310
    adafruit/Adafruit INA219
    adafruit/Adafruit HTS221
    adafruit/Adafruit HTU21DF Library
    adafruit/Adafruit HTU31D Library
    adafruit/Adafruit LTR390 Library
    adafruit/Adafruit LTR329 and LTR303
    adafruit/Adafruit PCT2075
    adafruit/Adafruit SCD30
    adafruit/Adafruit SGP30 Sensor
    adafruit/Adafruit SGP40 Sensor
    adafruit/Adafruit Si7021 Library
    adafruit/Adafruit VCNL4020 Library
    adafruit/Adafruit VCNL4040
    adafruit/Adafruit MCP9808 Library
    adafruit/Adafruit MCP9600 Library
    adafruit/Adafruit MPL115A2
    adafruit/Adafruit MPRLS Library
    adafruit/Adafruit MS8607
    adafruit/Adafruit TMP117
    adafruit/Adafruit TSL2591 Library
    adafruit/Adafruit_VL53L0X
    adafruit/Adafruit VL53L1X
    adafruit/Adafruit_VL6180X
    adafruit/Adafruit PM25 AQI Sensor
    adafruit/Adafruit VEML7700 Library
    adafruit/Adafruit LC709203F
    adafruit/Adafruit LPS2X
    adafruit/Adafruit LPS35HW
    adafruit/Adafruit seesaw Library
    adafruit/Adafruit BME680 Library
    adafruit/Adafruit MAX1704X
    adafruit/Adafruit ADT7410 Library
    adafruit/Adafruit GFX Library
    adafruit/Adafruit STMPE610
    adafruit/Adafruit TouchScreen
    adafruit/Adafruit MQTT Library
    bblanchon/ArduinoJson
    PaulStoffregen/OneWire
[... 221 lines cut ...]
    -DWIFICC=CYW43_COUNTRY_UK
    -DDEBUG_RP2040_WIRE
    -DDEBUG_RP2040_SPI
    -DDEBUG_RP2040_CORE
    -DDEBUG_RP2040_WIFI
    -DNDEBUG
    -DLWIP_DEBUG
    -DDEBUG_RP2040_PORT=Serial1
    -DDEBUG_RP2040_UART_1
    -DDEBUG_RP2040_UART=1
    -Og
    ; Enable debug stack protection
    -fstack-protector
    ; Enable Exceptions
    -DPIO_FRAMEWORK_ARDUINO_ENABLE_EXCEPTIONS
    ; Enable RTTI
    -DPIO_FRAMEWORK_ARDUINO_ENABLE_RTTI
    ; ; Enable default USB Stack of Pico SDK USB Stack with none of below usb options
    ; Adafruit TinyUSB
    -DUSE_TINYUSB
    ; ; No USB stack
    ; build_flags = -DPIO_FRAMEWORK_ARDUINO_NO_USB
    ; -DPIO_FRAMEWORK_ARDUINO_ENABLE_IPV6
    
```

commit c7ea0f38 (2024-04-16): fsHalt message improvements. Closes #545
commit 82984850 (2024-04-16): fsHalt message improvements. Closes #545
PR #570 (2024-04-19): fsHalt message improvements
Closes #545
closed 2024-04-19 by tyeth