adafruit/Adafruit_Wippersnapper_Arduino #546: UART PM2.5 sensor stops responding after 5-11minutes

tyeth, 2024-02-08:
**Describe the bug**
Originally running on **Feather ESP32v2** latest wippersnapper (and a few versions ago, maybe 60's), also tested on **feather s3 tft**,
the sensor is set to 30seconds polling period, and after approximately 5minutes of good readings it would stop sending any data back to IO and the serial says data not available. I also another day (not espv2 but s3) saw it repeatedly do it after 11minutes in a later testing session.
What's most annoying is it runs for 5-10minutes before happening, then says nothing except the
```cpp
 if (!_aqi->read(&_data)) {
       Serial.println("[UART, PM25] Data not available.");
```

**To Reproduce**
Steps to reproduce the behavior:
Add Uart PM2.5 PMS5003 to esp32v2 or s3 tft feathers on wippersnapper device pages
Start logging serial (add timestamps).
Come back in 11 minutes and see whats happened.

brentru, 2024-02-08:
I may try isolating that section out of WipperSnapper, and into an Arduino sketch to see if the UART bus still fails to poll new data from this sensor long-term.

tyeth, 2024-02-08:
serial debug log for a few runs with minor change if data unavailable (https://github.com/tyeth/Adafruit_Wippersnapper_Arduino/tree/uart-pm25-wip):
[COM33_2024_02_08.19.40.24.715.txt](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/files/14214050/COM33_2024_02_08.19.40.24.715.txt)

matching serials (use time from last one of these to jump in Com33 file):
[COM45_2024_02_08.19.40.28.427.txt](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/files/14214055/COM45_2024_02_08.19.40.28.427.txt)
[COM45_2024_02_08.19.40.30.543.txt](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/files/14214056/COM45_2024_02_08.19.40.30.543.txt)
[COM45_2024_02_08.19.40.46.568.txt](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/files/14214057/COM45_2024_02_08.19.40.46.568.txt)

[COM45_2024_02_08.19.45.43.221.txt](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/files/14214051/COM45_2024_02_08.19.45.43.221.txt)

tyeth, 2024-09-24:
Assigning to myself to retest this sensor on C6 and PicoW. 
There has been a commit to extend the delay in the library example before attempting to use the sensor to 3seconds.
We may need similar or to just accept the failed first reading attempts.

tyeth, 2025-01-07:
Have a look at this linked issue: https://github.com/adafruit/Adafruit_PM25AQI/issues/14
and the original report / PR: https://github.com/adafruit/Adafruit_PM25AQI/pull/10

and crucially Carters suggested fix: https://github.com/adafruit/Adafruit_PM25AQI/pull/10#issuecomment-915529497

tyeth, 2025-05-16:
Double check this is fixed, the implementation of the fix was re-written by Brent in v2 of PM25 AQI library.

closed 2025-08-19 by brentru