adafruit/Adafruit_Wippersnapper_Arduino #562: WipperSnapper firmware fails at network connection when loaded from MacOS but succeeds on Windows

BlitzCityDIY, 2024-04-02:
**Describe the bug**
WipperSnapper firmware fails at network connection when loaded from MacOS but succeeds on Windows.

***Arduino board***
ItsyBitsy ESP32

**To Reproduce**
Steps to reproduce the behavior:
1. Using the espwebtool, I erased the flash
2. Then I uploaded the Factory Reset bin
3. After the reset bin was running successfully, I went thru the WipperSnapper GUI in Chrome to load the firmware.
4. On the Connect! step, the yellow light flashes, blue flashes multiple times and then is solid blue

**Expected behavior**
When I follow these steps on Windows 10, the firmware loads properly and the device is added to my account. I'm able to add components as well.

**Which components are connected to your device**
N/A

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: MacOS Ventura 13.3 (M1)
 - Browser: Chrome
 - Version: Chrome 123.0/WipperSnapper v1.0.0-beta.78

**Additional context**
I tried taking the board with the firmware loaded via MacOS to my Windows 10 computer but it still hangs at the Connect! step (solid blue light)

Happy to test with other boards too if that's helpful.

brentru, 2024-04-03:
@tyeth  Do you have a laptop similar to Liz' that you can replicate this with? If you don't, I'll self assign and try to replicate.

tyeth, 2024-04-03:
@brentru I've an Intel Mac (not M1) running a recent beta of Sonoma (14.2 with an update to 14.5 available). Can attempt to initially recreate, but if you have an M1/2/3 or still on v13 then maybe that would be better

brentru, 2024-04-03:
@tyeth  I have a M1 on Ventura, self-assigned this to myself.

brentru, 2024-04-03:
> the only ones working are beta 72 and beta 66

videopixil, 2024-04-03:
I've been testing on windows 11

brentru, 2024-04-08:
@videopixil @BlitzCityDIY I followed your replication steps. However, I can't replicate this with a Macbook Pro running Ventura 13.6.3 and a QT PY ESP32 Pico. 

Follow-up Questions:
* Is this only with the ESP32 Pico, or other boards have the same issue?

* When you connect your ESP32 to the mac - could you open Arduino IDE's serial monitor? Then, press RESET to let the board attempt to connect. You should see serial output - could you copy that and paste it here?

BlitzCityDIY, 2024-04-08:
and here is the output from the same device after successfully loading the firmware with windows 10:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1256
load:0x40078000,len:13832
load:0x40080400,len:4
load:0x40080404,len:3048
entry 0x40080590
Adafruit.io WipperSnapper
-------Device Information-------
Firmware Version: 1.0.0-beta.78
Board ID: itsybitsy-esp32
Adafruit.io User: BlitzCityDIY
WiFi Network: wubbalubbadubdub
MAC Address: 0C:8B:95:96:BC:70
-------------------------------
ESP32 CPU0 RESET REASON: NO_MEAN
ESP32 CPU1 RESET REASON: POWERON_RESET
E (298) task_wdt: delete_entry(232): task not found
Generating device's MQTT topics...
Running Network FSM...
Establishing network connection...
Performing a WiFi scan for SSID...SSID found!
Connecting to WiFi (attempt #0)
Connected to WiFi!
Connecting to AIO MQTT (attempt #0)
WiFi Status: 20
Registering hardware with WipperSnapper...
Registering hardware with IO...
Encoding registration request...Encoding registration msg...Published!
Polling for registration message response...2
GOT Registration Response Message:
Hardware Response Msg:
	GPIO Pins: 22
	Analog Pins: 6
	Reference voltage: 3.30v
Completed registration process, configuration next!
Polling for message containing hardware configuration...
Polling for message containing hardware configuration...
Polling for message containing hardware configuration...
cbSignalTopic: New Msg on Signal Topic
2 bytes.
decodeSignalMsg
cbSignalMsg
Sub-messages found: 1
Signal Msg Tag: Pin Configuration
Initial Pin Configuration Complete!
Publishing to pin config complete...
Hardware configured successfully!
Registration and configuration complete!
Running application...
PING!
PING!
```

could it be at the IO account level? i could try having Noe login on my windows machine and vice versa

videopixil, 2024-04-08:
hi hi @brentru  i tried 
feather 32 s2 and s3
qtpy32 s2
pi pico w
itsybitsy 32

brentru, 2024-04-08:
@videopixil  It seems to be an issue with connecting to Adafruit IO, not the WiFi network. Can you verify that the adafruit io key for the `pixil3d` account matches what appears on io.adafruit.com? Make sure the same key is used on the devices.

videopixil, 2024-04-09:
@brentru  yes verified io keys match - the format is:
{
  "io_username": "pixil3d",
  "io_key": "same-IO-key",
  "network_type_wifi": {
    "network_ssid": "fake-network1",
    "network_password": "fakepass"
  },
  "status_pixel_brightness": 0.2
}

BlitzCityDIY, 2024-04-09:
it does seem to be something with the account. i signed into the Mac with my Adafruit account and was able to load the firmware with the same ItsyBitsy.

videopixil, 2024-04-24:
@brentru I regenerated the io key and it worked! - new devices are able to connect 🎉

brentru, 2024-04-25:
Weird, but glad it was account-centric. Closing for now, feel free to reopen if you have this issue occur again.

closed 2024-04-25 by brentru