adafruit/Adafruit_Wippersnapper_Arduino #566: Compilation Error

ericarjones, 2024-04-11:
**Describe the bug**
A clear and concise description of what the bug is.

***Arduino board***
Adafruit Feather M4 Express

**To Reproduce**
Steps to reproduce the behavior:
1. Open Arduino IDE
2. Go to the Wippersnapper_NoFS example under file > examples > Adafruit Wippersnapper > Wippersnapper_NoFS
3. Verify then upload sketch 

**Expected behavior**
Device should be detected by Wippersnapper .

**Which components are connected to your device**

Adafruit Feather M4 Express
Which components are set up on the WipperSnapper device page?

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
**Desktop (please complete the following information):**
 -iOS 
 - Browser: Safari 
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device:  Iphone 13
 - OS: 17.2.1
 - Browser: Safari
 - Version 17.2.1

**Additional context**
Add any other context about the problem here.

brentru, 2024-04-11:
@tyeth Would you have time to take a look at this report?

brentru, 2024-04-12:
@ericarjones  While the ATSAMD51 is supported by WipperSnapper, the Adafruit Feather M4 is not currently supported by WipperSnapper.

Are you following the guide and intend on adding this board to WipperSnapper?

brentru, 2025-03-03:
I am closing this issue due to inactivity. If you would like to comment on this issue again, please re-open it.

closed 2025-03-03 by brentru