adafruit/Adafruit_Wippersnapper_Arduino #579: unable to connect to mqtt from my featheresp32

mcsarge, 2024-05-04:
I am trying to setup my featheresp32 on IO and I am getting this:
-------Device Information-------
Firmware Version: 1.0.0-beta.78
Board ID: feather-esp32
Adafruit.io User: mcsarge
WiFi Network: SARGENET
MAC Address: 24:6F:28:02:9D:C8
-------------------------------
ESP32 CPU0 RESET REASON: NO_MEAN
ESP32 CPU1 RESET REASON: POWERON_RESET
E (65) task_wdt: delete_entry(232): task not found
Generating device's MQTT topics...
Running Network FSM...
Establishing network connection...
Performing a WiFi scan for SSID...SSID found!
Connecting to WiFi (attempt #0)
Connected to WiFi!
Connecting to AIO MQTT (attempt #0)
WiFi Status: 20
Unable to connect to Adafruit IO MQTT, retrying in 3 seconds...
Connecting to AIO MQTT (attempt #1)
WiFi Status: 20
Unable to connect to Adafruit IO MQTT, retrying in 3 seconds...
Connecting to AIO MQTT (attempt #2)
WiFi Status: 20
Unable to connect to Adafruit IO MQTT, retrying in 3 seconds...
Connecting to AIO MQTT (attempt #3)
WiFi Status: 20
Unable to connect to Adafruit IO MQTT, retrying in 3 seconds...
Connecting to AIO MQTT (attempt #4)
WiFi Status: 20
Unable to connect to Adafruit IO MQTT, retrying in 3 seconds...
ERROR [WDT RESET]: ERROR: Unable to connect to Adafruit.IO MQTT, rebooting soon...

brentru, 2024-05-06:
@tyeth  Do you want to take a look at this issue?

tyeth, 2024-05-07:
Yeah, I'll take an initial look. I'm going to create an issue to return the mqtt error response.

tyeth, 2024-05-07:
@mcsarge we've just released beta 82 so please give be that a try. You will probably still experience the issue, however the logging will now include the reason for the connection failure.
We've also updated the installer and partition layout for the featherESP32 so if the problem has disappeared then that could be an added benefit of these changes or moving to esp32-arduino v3.0.0-rc.1 along with the improvements to the 3rd party libraries we use in Wippersnapper.

Let us know how you get on...

mcsarge, 2024-05-08:
I loaded beta 82, got the same sort of failure, but there is more information in there on the error:

SSID found!
Connecting to WiFi (attempt #0)
E (19878) wifi:Set status to INIT
Connected to WiFi!
Connecting to AIO MQTT (attempt #0)
WiFi Status: 20
MQTT Connection Error: 5
Not authorized to connect
Unable to connect to Adafruit IO MQTT, retrying in 3 seconds...
Connecting to AIO MQTT (attempt #1)
WiFi Status: 20
MQTT Connection Error: -1
Connection failed
Unable to connect to Adafruit IO MQTT, retrying in 3 seconds...
Connecting to AIO MQTT (attempt #2)
WiFi Status: 20
MQTT Connection Error: 5
Not authorized to connect
Unable to connect to Adafruit IO MQTT, retrying in 3 seconds...
Connecting to AIO MQTT (attempt #3)
WiFi Status: 20
MQTT Connection Error: -1

brentru, 2024-05-08:
Thanks for the info, I'm going to go ahead and close this issue.

closed 2024-05-08 by brentru