adafruit/Adafruit_Wippersnapper_Arduino #586: Huzzah32-ESP32 goes offline after connecting Waterproof DS18B20

mennamorato, 2024-05-14:
**Describe the bug**
as title says, when i connect the Waterproof DS18B20 my feather does not report data and disconnects / reconnects / disconnects  etc from IO.   I have tried many different GPIO pins, check connections etc.

***Arduino board***
Huzzah32-ESP32 feather

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Add component'
2. Click on 'Waterproof DS18B20.'
3. Pick almost any pin for it
4. See it disconnect from Adafruit IO

**Expected behavior**
WAS working for over a few weeks then stopped working and cannot get it to work again
**Which components are connected to your device**
Which components are set up on the WipperSnapper device page?
AHT20 on I2C and Waterproof DS18B20 on different pins that I have tried

**Screenshots**
n/a

**Desktop (please complete the following information):**
 - OS: Mac OS X
 - Browser Safari
 - Version 17.4.1

**Additional context**
the funny thing is..it worked originally for a few weeks then stopped working and now I cannot get it to work again... I tried A0, A1, A2, D12, D27... maybe a few other pins.  I can leave it connected to the board, just remove it from the configuration then it stays connected.  So I do not think it's an electrical problem.  Also I have erased and refreshed whippersnapper binary a few times... no good

brentru, 2024-05-14:
@tyeth  Could you take a look at replicating this sometime this week?

tyeth, 2024-05-16:
Hey @mennamorato I've connected up what I believe is a similar setup, AHT20 on i2c (component added to device), and then added the DS18b20 (picked waterproof one but all the same), and used standard resolution (12bit) and pin A0 (D26).
Data appears to flow (at least the first data point) and not seen device reset yet.
[image]
[image]
[image]

My wiring has the DS18b20 and AHT20 powered from 3v pin. I have a 4.7k resistor between data and vcc pins of DS18b20, and the vcc pin is connected to 3v as mentioned rather than using parasitic power (and ground pin to ground).

If you haven't tried beta 82 via the IO web installer (new/upgrade device steps) then please do as it will erase your device fully along with installing the latest firmware.
I'll leave this running for a bit and check back tomorrow.

mennamorato, 2024-05-16:
Hi and thanks well… this happened to me yesterday .. i had it working on my bench for a few hours then it started to disconnect / reconnect and now does not work - on that pin.  its like when i move the pin it starts working until it doesn’t…
my wiring is same as yours
i will try beta 82 today 
thanks !

Mark

T-Mosher, 2024-05-16:
I'm curious, how are you physically doing the interconnects? Are you using one of the white plastic breadboards, or are you hard-wiring the pins?

I've found the DS one-wire sensors are extremely picky about intermittent connections, such as if you're using a breadboard and the contacts in a row are slightly worn down.

mennamorato, 2024-05-16:
white plastic breadboards…the Huzzah32 feather has headers installed then plugged into the breadboard BUT i have the waterproof sensor plugged into the header (female on top / male on bottom) right now. I did have it plugged into the breadboard which..could have been a connection intermittent issue? but would not expect the code to fail because of that i would expect no/bad data once in a while. the funny thing is..once it failed on pin A0 (for example) i could never get it to work on A0 again…if i kept moving it around (the data line that is) i would eventually find a pin for it to work

mark

T-Mosher, 2024-05-16:
_the funny thing is..once it failed on pin A0 (for example) i could never get it to work on A0 again…if i kept moving it around (the data line that is) i would eventually find a pin for it to work_

I've seen that behavior too, the drivers for the one-wire interfaces are not at all adaptable in real time. Any time I changed the wiring, I had to cycle the power and restart the code again.

Re: wiring. If you're using the bare part, then pushing the leads into the female header isn't recommended. You're relying on the lead diameter to match the size that the socket expects to receive. 

If you're using the waterproof sensor, then it has stranded wire leads, and again pushing those into a socket isn't recommended either.

Adafruit's breadboards aren't the highest quality, and I've worn out several rows by pushing headers into them more than just a few times. Intermittents are very difficult to debug.

mennamorato, 2024-05-16:
Ok cool....I ordered a different temp sensor (board) that is i2c.  I will
waterproof it myself with a thermally conductive electrically
non-conductive material and not use 1-wire which I never liked. Thanks for
the help.  Close this ticket.

tyeth, 2024-05-16:
Thanks Mark. Good luck and let us know if you have any other problems...