adafruit/Adafruit_Wippersnapper_Arduino #595: WS 1.0.0-beta.84 fails on Adafruit ESP32-S3 Feather with 4MB Flash 2MB PSRAM

microfoundry, 2024-06-27:
**Describe the bug**
I tried to update .83 to .84 and the device fails to boot afterwards, including a lack of COM port presenting to computer. Reverting back to .83 works as expected. 

***Arduino board***
Adafruit ESP32-S3 Feather with 4MB Flash 2MB PSRAM

**Which components are connected to your device**
STH45

**Desktop (please complete the following information):**
 - OS: Win11
 - Browser: chrome

brentru, 2024-06-27:
This issue is related to a bug within the ESP32 core release 3.0.2, which beta 84 is based on: https://github.com/espressif/arduino-esp32/issues/9936

brentru, 2024-06-27:
@microfoundry Until Espressif adds and releases a patch for this issue, we have rolled back WipperSnapper to Beta 83. Well cut a new release (beta 85) when the BSP is patched and ready.

brentru, 2024-06-27:
@microfoundry  We just released Beta 85 which should patch the issue you're experiencing with the Adafruit ESP32-S3 Feather w/4MB Flash 2MB PSRAM.

Please try this new firmware and let me know if the issue persists

microfoundry, 2024-06-28:
@brentru - Beta 85 loaded without issues. Good to go.

On a different topic, which just happened again before going to 85: If I have this same board running on battery and allow the battery to drain, the system will be toast. Removing the battery and connecting via USB only, I will not get a WIPPER drive nor connection to Adafruit IO. 2 for 2 times so far. Will test again...

brentru, 2024-06-28:
> On a different topic, which just happened again before going to 85: If I have this same board running on battery and allow the battery to drain, the system will be toast. Removing the battery and connecting via USB only, I will not get a WIPPER drive nor connection to Adafruit IO, but the uf2 bootloader is still available. 2 for 2 times so far. Will test again.

This is surprising to me. What battery and board are you using to test this?

microfoundry, 2024-06-28:
> What battery and board are you using to test this?

The same Adafruit ESP32-S3 Feather w/4MB Flash 2MB PSRAM with Adafruit SHT45 sensor. And a generic 300mA Li-Ion cell

tyeth, 2024-06-28:
Hey Terry, can you just confirm if that feather board has the MAX17048 or
the LC709203f lipo charger onboard (newer boards use the Max17048 but maybe
old stock so do check).
You can also just show the I2C address of the onboard battery charging
device (from the wippersnapper device page click I2C Scan button) as that
will equally tell us.

microfoundry, 2024-06-28:
Hi Tyeth - it's the LC709203f lipo charger.

tyeth, 2024-06-28:
Ah brilliant thank you for confirming a suspicion. We had a similar issue
with the MAX17048 and provided more generous error handling, but as there
were no reports of the lc709203f I believe it was left untouched.
Anyway that gives a good direction to head next week, so thanks again!

brentru, 2025-03-03:
@microfoundry  I am closing this stale issue due to inactivity. If you are still experiencing this functionality/behavior, please let us know and I'll re-open this issue to investigate.

closed 2025-03-03 by brentru