adafruit/Adafruit_Wippersnapper_Arduino #602: Increase the Potentiometer Component's frequency to fire more frequently

brentru, 2024-07-15:
**Is your feature request related to a problem? Please describe.**
> This is a feature request. I was really exicted to use Adafruit IO, but I ran into problems integrating a poteniometer for dimming a light. Return Interval: On Change leads to way too many requests that throttle adafruit io. Periodically every 30 seconds is way too slow. 

via https://forums.adafruit.com/viewtopic.php?t=212151

**Describe the solution you'd like**
> A wiresnapper poteniometer should have a Return Interval On Change - Debounce option.

**Describe alternatives you've considered**
It would also be possible to have it return every few detents, with a slight delay

@tyeth What are your thoughts?

tyeth, 2024-07-18:
Hmmm, 🤔 I wonder what's best. I know I made this problem more apparent when I adjusted the percentage before a change is registered (in line with an old comment in code).

In an ideal world for me I'd get notified of a change after it finished assuming it was relatively quick or a continuous movement, but maybe people want to see the granular change (reporting every x milliseconds until finished moving). In that instance id want a timer that would post a value in 200ms time based on old_value which would update if the movement continued and was above/below a threshold. Whether that's a sliding window based on more movement, or just post every x plus final window value if different, is the other matter.

The other thing is the threshold doesn't adjust logarithmically like one would want. We have fixed value (maybe 0.2% or 2% I forget) for reporting on_change, which on the specific Adafruit 10k potentiometer I tried was fine, but any larger value and we missed rotation at the top end of the nonlinear scale. I believe/assume different potentiometers exist with other behaviors (inverted or proportional). In an ideal worth we would have that as a setting for the wippersnapper analog components and the threshold would adjust nicely.

brentru, 2024-07-19:
> but maybe people want to see the granular change (reporting every x milliseconds until finished moving)
I think this is what the user wanted

> The other thing is the threshold doesn't adjust logarithmically like one would want.
Maybe we need to implement a more dynamic thresh. where smaller changes are not constantly reported. I agree with your timer approach, providing a fixed time window.

tyeth, 2024-11-15:
Initial release of changed behaviour in v1.0.0-beta.93, as part of #635.
Closing this issue for now, but please reopen if anyone has comments, or create a new issue.