adafruit/Adafruit_Wippersnapper_Arduino #609: Analog Input values are incredibly low on PICOW

tyeth, 2024-07-30:
A user reported to IO Support that their values for a 2.25v source on any analog input pin on the pico was minimal / unexpectedly small (0.3v).

Testing locally I see a value between 2 and 1023 being detected, and then scaled incorrectly. It returns 0.00 to 0.05.

Changing these lines to include setting a native resolution and scaleAnalogRead here appears to fix the issue (now 2 represents 0.06v and 1023 = 3.23v): [src/components/analogIO/Wippersnapper_AnalogIO.cpp#L84](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/blob/3a64cec38f41db3b24c9d15980b93434282f6fa4/src/components/analogIO/Wippersnapper_AnalogIO.cpp#L84)

I added to this if block along with a load of logging in branch  :
```
#if defined(ARDUINO_ARCH_SAMD)
  analogReadResolution(16);
  _nativeResolution = 12;
#elif defined(ARDUINO_ARCH_ESP32)
  scaleAnalogRead = true;
  _nativeResolution = 13;
#elif defined(ARDUINO_ARCH_RP2040)
  scaleAnalogRead = true;
  _nativeResolution = 10;
#else
  scaleAnalogRead = true;
  _nativeResolution = 10;
#endif
```

Tagging @brentru who has more of a grasp on this.

Also worth of note is the possibly related older issue #403

tyeth, 2024-07-30:
Added PR #610 to test the change

brentru, 2024-08-26:
@tyeth  another forum user for the analog read on pico w - https://forums.adafruit.com/viewtopic.php?t=212531

tyeth, 2024-09-18:
Closing this as appears to have been solved and released. Feel free to reopen or create a new issue if this still affects you

closed 2024-09-18 by tyeth