adafruit/Adafruit_Wippersnapper_Arduino #621: qt py loses secrets.json when battery goes down

migs35323, 2024-08-22:
**Describe the bug**

my qt-py s3 randomly loses the secrets.json credentials. when battery power goes out/down
i had noticed this issue in firmware beta .88 and right now i was testing with alpha .89 and same behavior
I usually power it through usb but i had been testing battery endurance, so i had been letting the battery go out on purpose.

as a side note, I once had the yellow light and boot out telling me the credentials where wrong, when i opened them in notepad/vscode, the file was correct. but i had to delete the file and recreate it to accept the credentials, however i can't seem to reproduce this again.
in this last behavior the only thing that comes to my mind, is there any sensitivity between LF and CRLF in the file? maybe i edited the file in windows and mixed it in ubuntu and WipperSnapper is not liking it...

***Arduino board***
qt-py s3 psram version aka qtpy-esp32s3-n4r2

**To Reproduce**
Steps to reproduce the behavior:
1. setup qt py with battery and best friend
2. let it run out of battery
3. when plugging it again it shows yellow light, boot.out warns about credentials, secrets.json is back to stock again.
PS: this doesn't happen every time, but about 2 out of 3 times it's not 100% consistent

**Expected behavior**
secrets,json to not be erased, 
or maybe turn it into a toggeable feature where we can choose the secrets to be erased as security feature lol

**Which components are connected to your device**
qt py s3 (psram version) with battery best friend attached + 2 adafruit temperature sensors connected with stemma qt, 

**Desktop (please complete the following information):**
 - OS: ubuntu 22 or windows 11

**Additional context**
Add any other context about the problem here.

brentru, 2024-08-22:
@tyeth  Do you have cycles available to look at this bug?

migs35323, 2024-08-22:
> @tyeth Do you have cycles available to look at this bug?

what do you mean by cycles? battery cycles?

brentru, 2024-08-22:
@migs35323  Tyeth is an Adafruit engineer who works on WipperSnapper with me. 

To clarify - I'm using "cycles" instead of "time", so I'm asking him if he has time to look at this or if I should.

tyeth, 2024-08-23:
👋 I've been running with shorter cycles  myself this week, medical issue, so likely to have a look around the middle of next week (once BFF arrives).
 It's possible that using a mix of line endings might be considered corrupt and it self formatting. At the same time I have seen similar behaviour before. Will report back.

tyeth, 2024-08-27:
Hardware just arrived, will setup a test

migs35323, 2024-08-28:
> Hardware just arrived, will setup a test. Not that it should matter, but please comment as to which sensors you have attached to the StemmaQT port, and which kind of battery

it's a [pimoroni galleon battery ](https://shop.pimoroni.com/products/galleon-400mah-battery?variant=40061068673107)(really love that one, it has a plastic shell so it doesn't freak me that i might pierce it with a header pin by accident and burn my house down lol)
i usually have some sensors like the tmp117, SHT45 and now MCP9808, i noticed it doesn't matter which ones i have..

I also have it connected usually to my raspberry pi 4 usb port as a power source, its a data cable but the rpi is not automouting it..
but i noticed it doesn't matter as well, happens no matter where i power it

i noticed that the file basiscally reverts to "insert credentials"
so it doesn't get like blank or have some weird characters, i edit it, save it, the qt py s3 connects to ada io, but when power goes out, i connect it to a usb port and YELLOW LIGHT, file reverts back to original "PLS FILL ME" state

tyeth, 2024-10-11:
So to keep you in the loop, this is being looked at, and I did manage to recreate the situation with enough cycles of the device.
We want deep sleep / offline logging as a long term aim, and this is a showstopper.
 I've also seen the file not in sync until the device was rebooted (like on the first boot the save was ignored), and I now superstitiously delete the secrets file before replacing it with a previously downloaded one. It's not consistent at all, and was for me on Windows, and as you say after doing a reboot (i.e. on second ever boot) it's resolved. That may need to be broken out into a separate issue.

Timeline8, 2024-12-20:
I opened a post in the forums for this same issue, not realizing this was already reported. My post with details is here https://forums.adafruit.com/viewtopic.php?p=1039724 but the short version is Qt Py S3, lipo charging BFF, a non-Adafruit 850mAh lipo, and a non-Adafruit AHT20 connected via the stemma connector, I2C, and taking reading every 15 minutes. I let it run all the way down and both times I lost the secrets file as well the first time all the previous CircuitPython stuff on the board before I switched over to Wippersnapper. I run a Mac when connecting to the board if that matters.

brentru, 2024-12-23:
@Timeline8  You may be able to test out the artifact from the most recent build of #656, which fixes the brownout issue you and Tyeth are having. We are likely not going to merge anything in and release until we are fully back, post-Holidays.

https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/actions/runs/11924340659/artifacts/2210637190

Timeline8, 2024-12-23:
@brentru, Thanks I will give it a try and report back. I did another test today with a different smaller battery and the same result. 3 for 3 losing the secrets file. So happy to try out the new build. Thanks again.