adafruit/Adafruit_Wippersnapper_Arduino #627: Data recorded in stuttering manner with SCD-30 and many boards (QT PY S2, QT PY S3, etc) - in Release 0.88

kcowin, 2024-09-09:
**Describe the bug**
A clear and concise description of what the bug is.

SCD-30 sensor has four feeds that can be accessed. When sending the feeds
to wippersnapper every 30 seconds, they show up randomly.

[image]

**To Reproduce**
Steps to reproduce the behavior:

Attach an SCD-30 to a QT PY Board and see if you can get better results.

**Expected behavior**
You will likely see missing values, if any at all.

**Which components are connected to your device**
Which components are set up on the WipperSnapper device page?

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Problem appears regardless of browser or browsing platform. There are no data errors
on the wippersnapper subscription page.

tyeth, 2024-09-19:
Provisional fix in PR. @kcowin if you wish to try the build assets they are attached 
[issue632-qtpy-assets.zip](https://github.com/user-attachments/files/17063533/issue632-qtpy-assets.zip)

tyeth, 2024-09-23:
I've been running it for a couple of days, at 30second intervals so about 3500(3467) readings. 

The temp/humidity get read in one command, and then the CO2 afterwards (so more chance of a failure).

Approximately 35 CO2 readings (1%) were more that 30seconds apart, the majority at 60s apart (so a reading was skipped), a sprinkle at 90s (two readings skipped), and a few further apart (presumably connectivity issues or account throttle).

Temperature readings had 10 entries out of 3500 that didn't fall in the 30second poll period.

[image]

kcowin, 2024-09-23:
Yes- I noticed it was not perfect, but a substantial improvement, and now the data is usable. KenSent from my iPhone

image.png (view on web)

closed 2024-10-11 by tyeth