adafruit/Adafruit_Wippersnapper_Arduino #628: Beta 89 does not connect board to IO service and restarts board every minute

Timeline8, 2024-09-11:
**Describe the bug**
Beta 89 was released to address an issue with the WIPPER board not being shown. However Beta 89 does not work. Boards loaded with 89 do not connect to the IO service with the Secrets files properly populated and while the WIPPER drive does show up now, boards continuously disconnect and reconnect to the computer about once a minute.

***Arduino board***
I found this to be true on both a Qt Py S3 and a Feather ESP32-S3 Reverse TFT.
In addition this has now been reported two more times by other users on the Adafruit forums...
https://forums.adafruit.com/viewtopic.php?t=213405 (Qt Py S2)
https://forums.adafruit.com/viewtopic.php?p=1028596 (Funhouse)

**To Reproduce**
See previous discussion on issue #591 
https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/issues/591

**Expected behavior**
I expect the board to not reboot over and over and to be recognized by the IO service.

**Which components are connected to your device**
Nothing else.

**Desktop (please complete the following information):**
Mac OS - Not at my Mac at the moment so can not fetch the specific details.

brentru, 2024-09-11:
> Boards loaded with 89 do not connect to the IO service with the Secrets files properly populated and while the WIPPER drive does show up now, boards continuously disconnect and reconnect to the computer about once a minute.

@Timeline8  Are you able to obtain the output from the serial monitor when this occurs? What does the boot text file on the WIPPER drive say?

Timeline8, 2024-09-11:
I will try to obtain both those this evening when I am back at home. Stay tuned.

Timeline8, 2024-09-11:
Interesting. Says it can't find WiFi network.

Boot text as follows:
`Adafruit.io WipperSnapper
Firmware Version: 1.0.0-beta.89
Board ID: feather-esp32s3-reverse-tft
MAC Address: 34:B7:DA:58:87:CC
ESP-IDF Version: v5.3-369-g466a392a76-dirty
ESP32 Core Version: 3.1.0`

and serial output is `ERROR [WDT RESET]: ERROR: Unable to find WiFi network, rebooting soon...` so that explains the regular reboot. However if I did nothing other than put in Beta 85 and restarted the board (I didn't touch any files at all) when I go to the IO page it recognized the board immediately so clearly my secrets files is correct. '85 seems to not be using the WiFi correctly.

brentru, 2024-09-12:
Another issue: https://forums.adafruit.com/viewtopic.php?t=213422

brentru, 2024-09-12:
I've done a bit of preliminary investigation. The root cause of the bug is not directly related to WipperSnapper, but is caused by the board support package (BSP) we use for ESP32 devices. Specifically, the latest release of the arduino-esp32 BSP (v3.1.0-RC1) has introduced a bug that breaks WiFi scanning functionality. ESP32-x boards running WipperSnapper Beta 89 use this BSP version.

In WipperSnapper, we perform a scan for the WiFi network specified in secrets.json. If the network is not found - the device reboots. As a result, WipperSnapper devices are repeatedly detaching and reattaching because they are unable to successfully find the WiFi Access Point (AP) due to the broken WiFi scanning in the BSP.

Related issues:
[Issue #1 in Arduino-ESP32](https://github.com/espressif/arduino-esp32/issues/10281)
[Issue #2 in ESP-IDF](https://github.com/espressif/esp-idf/issues/14505)

While a patch (https://github.com/espressif/arduino-esp32/pull/10282) has been proposed to address this issue, espressif releases a new version of the BSP on their schedule and we do not have influence over when they release.

Today, I'll look at rebuilding WipperSnapper on a more stable BSP and try to release Beta 90, fixing this WiFi scanning issue.

Timeline8, 2024-09-12:
This is why we leave it to you wicked sm'aht programming types to figure this stuff out for us. ;)  I am confident you will have a solution soon and I will be happy to give it a test. 

I am in no hurry personally since I use the IO service through the CircuitPython library, but I am sure there are people who want to give wippersnapper a try and they are confused when they can not get it to work. Good luck with the rebuild.

brentru, 2024-09-12:
Thanks for providing me with enough information to help figure it out. I am in the middle of WipperSnapper 1.0.0 and noticed this happening yesterday but chalked it up to the non-beta version until I saw others were having the same issue.

TD-er, 2024-09-12:
You can just allocate the struct and completely wipe it yourself before using it.
This will fix these issues and you don't need to wait for new releases.

TD-er, 2024-09-12:
Problem is the newly introduced channel bitmap struct is not properly initialized.
When set to all zeroes, it will scan again all channels or the given channel.

So just to be sure you're not missing anything, it is best to completely memzero all of the struct.
The IDF example code does this by using `calloc` to allocated a chunk of memory the size of this struct and since calloc clears the memory, this error will not occur.
But if you create this struct on the stack you must clear it yourself.

brentru, 2024-09-12:
@Timeline8  I've released Beta 90, please give it a try! 

@TD-er  Thank you for your help and for the clarifications. I'm going to move around the build toolchain so we can quickly patch arduino-esp32 bsp and move without relying on releases.

closed 2024-09-12 by brentru

Timeline8, 2024-09-12:
Yep, I can confirm that now my S3 Reverse TFT Feather running beta 90 works. I power cycles the board and still see the drive and I was able to hook right up to IO share, pressed buttons, played with the NeoPixel, ran my data rate over the 30 of 30 for my free account... OOPS... hope none of my boards sending data to their feeds don't crash.

Nice job, Brent.

brentru, 2024-09-13:
> played with the NeoPixel, ran my data rate over the 30 of 30 for my free account... OOPS... hope none of my boards sending data to their feeds don't crash.

Awesome, glad it's working for you! 

> Now where do I send my suggestions for improving the IO service web page as I noticed that it is kind of annoying to add a board when it is already detected and you don't need the step by step instructions. I assume this is a website thing?

Go to https://io.adafruit.com/support -> Contact Support -> "I have a suggestion for Adafruit IO"

Timeline8, 2024-09-13:
Thanks, I just put in my two cents worth of a suggestion.