adafruit/Adafruit_Wippersnapper_Arduino #630: WipperSnapper on Funhouse causes bootloop and crash

brentru, 2024-09-12:
**Describe the bug**
From (https://forums.adafruit.com/viewtopic.php?p=1028567#p1028567) but also observed by me: 

> The screen on the Funhouse appears corrupted and pixelated for a brief moment, and the lights flash quickly (including the red LED near the reset button) before the cycle repeats, all within 3 seconds. This behavior continues until you power off the device.

***Arduino board***
Adafruit FunHouse ESP32-S2

**To Reproduce**
Install WipperSnapper Beta on FunHouse via io.adafruit.com's web uploader

**Expected behavior**
TFT should come into the loading screen, not crash

**Which components are connected to your device**
None

OS Tested: MacOS, Windows 10

**Smartphone (please complete the following information):**
n/a

**Additional context**
Forum post with original problem: https://forums.adafruit.com/viewtopic.php?p=1028567#p1028567

brentru, 2024-09-13:
@tyeth  I spent an hour or so this morning tracking this issue down, you may have more experience with LVGL and want to try taking a look next week:

Where this issue is originating from:
```LvGLStatus Adafruit_LvGL_Glue::begin(Adafruit_SPITFT *tft, void *touch,bool debug)``` calls `lv_init()` (LVGL library) which causes a crash. LVGL_Glue does not make it past the `begin()` call. 

Challenges while debugging, on my end:
* My ESP32 doesn't seem to get the "Guru Meditation Error" that the related issue got, ended up being difficult to track where I am in the code
* It doesn't seem lv_conf.h is being included on my build system (Arduino IDE 2.x), so I can't get detailed logs of what's occurring within `lv_init()`. Possibly this is the same for you and you'll want to fix this first to get insight into `lv_init()`.

Related Issue: https://github.com/adafruit/Adafruit_LvGL_Glue/issues/23

tyeth, 2024-09-19:
v3.0.5 was released this week, and building with that in platformIO / vscode I don't see the bug (display initialises and board comes online).
I used the same library versions of lvgl + littlevgl glue as the CI script installs. 

Had issues building in arduino ide, will revisit tomorrow

tyeth, 2024-09-20:
Got it building in arduino ide (BSP 3.0.5). Appears to respect the lv_conf.h I've added to my sketch folder. Runs fine.
Recreated the bug on 3.1.0-rc.0. (Been testing WS v88 the whole time). Running with trace output isn't showing anything obvious

tyeth, 2024-09-20:
log attached 
[Funhouse_lvgl_reboot_bsp310RC0.txt](https://github.com/user-attachments/files/17076435/Funhouse_lvgl_reboot_bsp310RC0.txt)

brentru, 2024-09-23:
@tyeth  I'll try this today, you think it's isolated to BSP 3.1.0-rc.0?

tyeth, 2024-09-23:
It appeared to be, as I got a build out of 3.0.5, but I wouldn't bet the house on it.

tyeth, 2024-10-11:
@brentru I think that was part of the reason for pinning the ci-arduino repo (ci-wippersnapper branch).
I haven't retested recently, but as we're pinned at 3.0.5 all should be well.
No code was changed as part of a fix, just tooling.

brentru, 2024-10-11:
Right Okay, going to close this out

tyeth, 2024-10-21:
Closed with the release of v1.0.0-beta.91 (retested and same funhouse is now happy)

closed 2024-10-21 by tyeth

tyeth, 2025-01-27:
Note: this is still pinned to BSP < 3.1.x for FunHouse / LvGL Glue