adafruit/Adafruit_Wippersnapper_Arduino #636: C6 build in CI causes boot loop with "ets_loader.c 67" message. Local arduino-cli builds work fine. (Both with BSP 3.0.5 + 3.1.0-rc.1)

tyeth, 2024-10-03:
Adafruit Feather C6 builds from CI jobs are causing a boot loop, with a message similar to:

tyeth, 2024-10-03:
Tagging @brentru to take a look later

tyeth, 2024-10-03:
Related discussion (nothing so far): https://github.com/espressif/arduino-esp32/discussions/10418

brentru, 2024-10-18:
Here is my build of `main` 
```
13:56:45.504 -> E (1239) esp_littlefs: ./managed_components/joltwallet__littlefs/src/littlefs/lfs.c:1369:error: Corrupted dir pair at {0x0, 0x1}
13:56:45.504 -> 
13:56:45.504 -> E (1240) esp_littlefs: mount failed,  (-84)
13:56:45.504 -> E (1244) esp_littlefs: Failed to initialize LittleFS
```

Dependency Log:
```
Using library Adafruit WipperSnapper at version 1.0.0-beta.90 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_Wippersnapper_Arduino 
Using library Adafruit MQTT Library at version 2.5.8 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_MQTT_Library 
Using library Adafruit SleepyDog Library at version 1.6.5 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_SleepyDog 
Using library SPI at version 3.0.5 in folder: /Users/brentrubell/Library/Arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/SPI 
Using library Adafruit DotStar at version 1.2.3 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_DotStar 
Using library Adafruit BusIO at version 1.16.1 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_BusIO 
Using library Adafruit NeoPixel at version 1.12.3 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_NeoPixel 
Using library ArduinoJson at version 7.0.4 in folder: /Users/brentrubell/Documents/Arduino/libraries/ArduinoJson 
Using library Wire at version 3.0.5 in folder: /Users/brentrubell/Library/Arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/Wire 
Using library Adafruit Unified Sensor at version 1.1.12 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_Sensor 
Using library Adafruit ADT7410 Library at version 1.2.0 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_ADT7410_Library 
Using library Adafruit AHTX0 at version 2.0.5 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_AHTX0 
Using library hp_BH1750 at version 1.0.2 in folder: /Users/brentrubell/Documents/Arduino/libraries/hp_BH1750 
Using library Adafruit BME280 Library at version 2.2.2 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_BME280_Library 
Using library Adafruit BME680 Library at version 2.0.2 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_BME680_Library 
Using library Adafruit BMP280 Library at version 2.6.8 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_BMP280_Library 
Using library Adafruit BMP3XX Library at version 2.1.2 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_BMP3XX_Library 
Using library Adafruit DPS310 at version 1.1.1 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_DPS310 
Using library Adafruit DS248x at version 1.1.0 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_DS248x 
Using library ENS160 - Adafruit Fork at version 3.0.1 in folder: /Users/brentrubell/Documents/Arduino/libraries/ENS160_-_Adafruit_Fork 
Using library Adafruit HDC302x at version 1.0.1 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_HDC302x 
Using library Adafruit HTS221 at version 0.1.2 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_HTS221 
Using library Adafruit HTU21DF Library at version 1.1.2 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_HTU21DF_Library 
Using library Adafruit HTU31D Library at version 1.1.3 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_HTU31D_Library 
Using library Adafruit INA219 at version 1.2.1 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_INA219 
Using library Adafruit LC709203F at version 1.3.2 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_LC709203F 
Using library Adafruit LPS2X at version 2.0.4 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_LPS2X 
Using library Adafruit LPS35HW at version 1.0.4 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_LPS35HW 
Using library Adafruit LTR329 and LTR303 at version 2.0.1 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_LTR329_and_LTR303 
Using library Adafruit LTR390 Library at version 1.1.2 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_LTR390_Library 
Using library Adafruit MAX1704X at version 1.0.2 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_MAX1704X 
Using library Adafruit MCP3421 at version 1.0.0 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_MCP3421 
Using library Adafruit MCP9808 Library at version 2.0.0 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_MCP9808_Library 
Using library Adafruit MPL115A2 at version 2.0.0 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_MPL115A2 
Using library Adafruit MPRLS Library at version 1.2.0 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_MPRLS_Library 
Using library Adafruit MS8607 at version 1.0.2 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_MS8607 
Using library Adafruit NAU7802 Library at version 1.0.4 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_NAU7802_Library 
Using library Adafruit PCT2075 at version 1.0.3 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_PCT2075 
Using library Adafruit PM25 AQI Sensor at version 1.1.0 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_PM25AQI 
[... cut ...] 237A774E8A5676C87F67D8D568BBA/Wippersnapper_demo.ino.bootloader.bin" 0x8000 "/private/var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino/sketches/8D5237A774E8A5676C87F67D8D568BBA/Wippersnapper_demo.ino.partitions.bin" 0xe000 "/Users/brentrubell/Library/Arduino15/packages/esp32/hardware/esp32/3.0.5/tools/partitions/boot_app0.bin" 0x10000 "/private/var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino/sketches/8D5237A774E8A5676C87F67D8D568BBA/Wippersnapper_demo.ino.bin" 
esptool.py v4.6
```

PR #651 (2024-10-22): Fix C6 Build
This pull request updates to the `.github/workflows/build-clang-doxy.yml` file to resolve the bootloop for the ESP32-C6 
build.

Repository and dependency updates:
* Changed the repository reference from `tyeth/ci-arduino` to `adafruit/ci-arduino` and updated the branch reference to `ci-wippersnapper` (`.github/workflows/build-clang-doxy.yml`).
* Updated the `pip` command to install a specific version of `esptool` (`esptool==4.6`) instead of the latest version (`.github/workflows/build-clang-doxy.yml`).

Resolves: #636
closed 2024-10-22 by brentru