adafruit/Adafruit_Wippersnapper_Arduino #657: Wippersnapper 0.92 breaks SCD30 data feed to Adafruit QT Py S3 (2MB PSRAM) - and probably other boards

kcowin, 2024-11-04:
**Describe the bug**
A clear and concise description of what the bug is.

Wippersnapper 0.92 breaks SCD30 data feed to Adafruit QT Py S3 (2MB PSRAM) - and probably other boards

***Arduino board***
INSERT ARDUINO BOARD NAME/TYPE HERE

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

I expect the data to be representative of what the sensor experiences.
Instead it is garbled.

**Which components are connected to your device**
Which components are set up on the WipperSnapper device page?

**Screenshots**
If applicable, add screenshots to help explain your problem.

Screenshot attached

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.
[image]

brentru, 2025-03-03:
Assigning myself to see if this is still happening on Beta 97

tyeth, 2025-03-07:
Should be fixed in #709