adafruit/Adafruit_Wippersnapper_Arduino #659: Pico (and other boards) connect to the first matching SSID, instead of picking the least negative RSSI value

tyeth, 2024-11-08:
I've got a mesh network with 3 nodes (plus base station router), all under the same SSID.
The pico detects one at -82db (over ten feet away) and fails to connect to it usually, instead only successfully connecting when it connects to the mesh with -32db (2feet away).

In an ideal world we would evaluate the entire network list from the scan / scan result, or maybe just order by RSSI first then reusing existing logic.

tyeth, 2024-12-12:
Watchdog timer gets in the way of wifi Multi class (it's not watchdog aware) for Pico. This doesn't present unless using alternative_networks, another spanner in the works. Watchdog cannot be disabled on pico once enabled, so the only option is to not enable it in the first place until the wifi has connected (small risk of device not rebooting in major error like failed to generate device uid, but not rebooting wont matter then). Assuming WDT is on and wifi fails then will attempt reconnect (and assuming over 8seconds will reset and try again without watchdog timer, or if connects then no issue)

brentru, 2025-03-03:
Resolved by #661