adafruit/Adafruit_Wippersnapper_Arduino #677: DS18b20 and Pico W causing decode error

brentru, 2025-01-06:
**Describe the bug**
Pico W connected to a ds18b20 is causing a crash

***Arduino board***
Raspberry Pi Pico W

**Logs**

Connecting a new DS18b20 to a Pico W, attempt #1
```
* NEW MESSAGE [Topic: Signal-DS]:
23 bytes.
[Message Type] Init. DS Sensor
Adding DS18x20 Component...Failed to find DSx sensor on specified pin.
Created OneWireBus on GPIO D14 with DS18x20 attached!
-> DS18x Init Response...Published!
Protobuf decode error: (none)
ERROR: Unable to decode DS message
```

Connecting a new DS18b20 to a Pico W, attempt #2 
```
* NEW MESSAGE [Topic: Signal-DS]:
23 bytes.
[Message Type] Init. DS Sensor
Connection lost -- read failed: [Errno 6] Device not configured
Use Stop/Restart to reconnect.
Process ended with exit code 1.
MQTT Connection Error: -1
Connection failed
...
ERROR [WDT RESET IN -1]:
ERROR: Unable to connect to Adafruit.IO MQTT, rebooting soon...
#Pico rebooted and same message repeated over and over#
```

@tyeth  I think this is from within the DS18 driver used on PicoW?
```
Connection lost -- read failed: [Errno 6] Device not configured
Use Stop/Restart to reconnect.
```

**Which components are connected to your device**
DS18b20 from Aliexpress

brentru, 2025-01-06:
@tyeth  Could you attempt to reproduce this issue this week?

tyeth, 2025-01-06:
Will do!

tyeth, 2025-01-27:
Just to add we may have a destructor missing issue, which presents itself when removing and adding DS18b20 devices.
Adding and removing the ds18b20 crashes on third time. Adding a PIO destructor fixes it, see 
https://github.com/tyeth/pstolarz_OneWireNg/tree/pio-destructor
and 
https://github.com/pstolarz/OneWireNg/issues/71

Note to self, retest with fix in https://github.com/pstolarz/OneWireNg/issues/71

pstolarz, 2025-01-28:
I don't recommend the bit-bang driver since it blocks interrupts during bus transmission. There are observed UART problems with PicoSDK while using the BB driver. While properly configured the PIO driver supports up to 8 object services - details how to created them will be available in the updated OneWireNg's README soon.

tyeth, 2025-01-28:
Retested the destructor PR and works, resolves the inability to re-add DS18b20 devices to the pico.

I haven't seen the not connected / connection lost read message, so not sure if the original bug was truly recreated rather than a second symptom / bug fixed.

brentru, 2025-01-31:
@tyeth  If it works, is there a corresponding PR for this?

brentru, 2025-02-21:
@tyeth  Can we close this?

tyeth, 2025-02-21:
I was kind of hoping to test with more than two DS18b20's, but I can only
locate two currently. I think there may be issues at 5 (or 3, basically 2
per pio channel is supported) which we would like to know about in advance.

brentru, 2025-03-03:
@pstolarz 
>  While properly configured the PIO driver supports up to 8 object services - details how to create them will be available in the updated OneWireNg's README soon.

Thanks for the feedback! Do you have an ETA of when to expect usage for the PIO driver in the readme? I'll gladly switch to it, especially if it supports 8 objects (that's awesome work, thank you!).

pstolarz, 2025-03-03:
@brentru It's there for some time already, see:
https://github.com/pstolarz/OneWireNg?tab=readme-ov-file#arch_rp2040pio_multi_bus

brentru, 2025-03-05:
@tyeth  So we migrated to the OneWireNG implementation instead of DallasTemperature in the v2 branch for WipperSnapper, not v1.

As a result, we won't fix this within v1 but it will be resolved within the v2 branch.

brentru, 2025-03-06:
@tyeth  The OneWireNG PIO implementation for RP2040 has been implemented in API-v2: https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/commit/09095e56f3a9ddbe0ff3dc3e3f3bcc57c9cc7668

Marking this as a WONTFIX for API-v1

closed 2025-03-06 by brentru