adafruit/Adafruit_Wippersnapper_Arduino #684: NeoPixels don't set to off when component removed or strand length updated

tyeth, 2025-01-23:
Changing the length of a neopixel component or removing it doesn't turn off the unused pixels. 
We should probably send a fill of off, then if needed set the new pixel values.

brentru, 2025-01-23:
Is this something that should be implemented on the firmware-side or the IO-side?

tyeth, 2025-01-25:
In my head we're getting an update message (or maybe remove then add), but we only set a new value for colour when a new colour is sent. If the user changes number then we should copy the current brightness and colour after quickly setting all to 0 to cover any now unused pixels.
It was noticeable when adding a second chain of 240 to an existing one of 256

tyeth, 2025-03-04:
Worth mentioning that when testing I added the command to turn off the leds in the destructor. 
https://github.com/tyeth/Adafruit_NeoPixel/tree/c6_destructor
This doesn't quite feel safe although might be, as I'm unsure if there is an expectation that pixels should keep running, but presumably not as in theory NeoPixel object maintains those running peripherals/timers.

closed 2025-04-23 by brentru