adafruit/Adafruit_Wippersnapper_Arduino #691: C6 Feather can't use onboard neopixel as user component

tyeth, 2025-01-31:
While doing the guide for Neopixel usage on the C6, it was noticed that the status illuminations work, but using the IO9 pin as a user component doesn't.

Serial:
```
16:25:54:683 -> Polling for message containing hardware configuration...
16:25:54:696 -> cbSignalTopic: New Msg on Signal Topic
16:25:54:696 -> 2 bytes.
16:25:54:696 -> decodeSignalMsg
16:25:54:696 -> cbSignalMsg
16:25:54:696 -> Sub-messages found: 1
16:25:54:696 -> Signal Msg Tag: Pin Configuration
16:25:54:696 -> Initial Pin Configuration Complete!
16:25:54:696 -> Publishing to pin config complete...
16:25:54:699 -> * NEW MESSAGE [Topic: Pixels]: 
16:25:54:699 -> 15 bytes.
16:25:54:699 -> [Message Type]: wippersnapper_signal_v1_PixelsRequest_req_pixels_create_tag
16:25:54:700 -> [ 22690][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x4204b044
16:25:54:700 -> [ 22691][V][esp32-hal-rmt.c:142] _rmtDetachBus(): Detaching RMT GPIO Bus
16:25:54:701 -> [ 22691][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 9 successfully set to type INIT (0) with bus 0x0
16:25:54:702 -> [ 22692][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 9 successfully set to type GPIO (1) with bus 0xa
16:25:54:702 -> [ 22693][E][esp32-hal-rmt.c:128] _rmtGetBus(): ==>_rmtWrite():GPIO 9 is not attached to an RMT channel.
16:25:54:703 -> Created NeoPixel strand of length 1 on GPIO #D9
16:25:54:703 -> -> wippersnapper_signal_v1_PixelsResponse...* NEW MESSAGE [Topic: Pixels]: 
16:25:54:706 -> 13 bytes.
16:25:54:706 -> [Message Type]: wippersnapper_signal_v1_PixelsRequest_req_pixels_write_tag
16:25:54:706 -> Filling color: 16777215
16:25:54:707 -> [ 22697][E][esp32-hal-rmt.c:128] _rmtGetBus(): ==>_rmtWrite():GPIO 9 is not attached to an RMT channel.
16:25:54:887 -> Published!
16:25:54:888 -> Hardware configured successfully!
16:25:56:448 -> Registration and configuration complete!
16:25:56:448 -> Running application...
```

tyeth, 2025-01-31:
```
16:56:53:079 -> Checking if status pixel is using GPIO #9
16:56:53:079 ->  = 9
16:56:53:079 -> lockStatusNeoPixel: 1
16:56:53:079 -> Releasing status NeoPixel
16:56:53:079 -> [ 14754][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x4204b0e4
16:56:53:080 -> [ 14755][V][esp32-hal-rmt.c:142] _rmtDetachBus(): Detaching RMT GPIO Bus
16:56:53:080 -> [ 14755][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 9 successfully set to type INIT (0) with bus 0x0
16:56:53:081 -> [ 14756][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 9 successfully set to type GPIO (1) with bus 0xa
16:56:53:081 -> deleted status NeoPixel
16:56:53:081 -> unlocked status NeoPixel
16:56:53:082 -> [ 14757][E][esp32-hal-rmt.c:128] _rmtGetBus(): ==>_rmtWrite():GPIO 9 is not attached to an RMT channel.
16:56:53:082 -> Created NeoPixel strand of length 1 on GPIO #D9
16:56:53:083 -> -> wippersnapper_signal_v1_PixelsResponse...* NEW MESSAGE [Topic: Pixels]: 
16:56:53:086 -> 13 bytes.
16:56:53:086 -> [Message Type]: wippersnapper_signal_v1_PixelsRequest_req_pixels_write_tag
16:56:53:086 -> Filling color: 16777215
16:56:53:086 -> [ 14761][E][esp32-hal-rmt.c:128] _rmtGetBus(): ==>_rmtWrite():GPIO 9 is not attached to an RMT channel.
```

tyeth, 2025-01-31:
It's not flowing as expected. When the neopixel object (status LED) is deleted, it doesn't reset the static rmtPin in esp.c part of NeoPixel.

If I set the destructor to be lower in the file (after declaring espShow), and add something like this to clear the old channel, then it works as expected:

```

/*!
  @brief   Deallocate Adafruit_NeoPixel object, set data pin back to INPUT.
*/
Adafruit_NeoPixel::~Adafruit_NeoPixel() {
  free(pixels);
  if (pin >= 0)
    pinMode(pin, INPUT);
#if defined(ESP32)
#if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 0, 0)
    if (begun)
    {
      log_d("Deinit RMT on pin %d (destructor)", pin);
      espShow(pin, NULL, 0, is800KHz);
    }
#endif
#endif
}
```

tyeth, 2025-01-31:
https://github.com/adafruit/Adafruit_NeoPixel/compare/master...tyeth:Adafruit_NeoPixel:c6_destructor

brentru, 2025-01-31:
Does this need a PR into `Adafruit_NeoPixel`?

brentru, 2025-03-03:
@tyeth  Circling back - does this still need to be added to Adafruit_NeoPixel? Do you want me to look at it this week, or do you have it?