adafruit/Adafruit_Wippersnapper_Arduino #701: Shadow component blocks after changing which subcomponents are enabled from a sensor

tyeth, 2025-02-22:
Connect new sensor with multiple subcomponents, only enable one of them. 
wait for reading.
Update sensor to have all subcomponents (I also turned off the first enabled one).
Data arrives for new options like expected, but sensor data/component blocks remain shadow blocks

[image]

brentru, 2025-03-03:
This seems like an issue on IO, rather than in WipperSnapper. Is that the case?

tyeth, 2025-03-07:
I need more info on it. Seemingly worse on staging although the same in theory in prod. I had to delete a device to clear up the situation at one point again recently. 
I think the situation may be two-fold, we are not responding to a message correctly (publish response failure), and then io-rails stalls the subcomponent (or sometimes seemingly all further broker to device UI initiated messages - maybe i2c only) without that failed message response.
 I've yet to properly investigate, but that was one reason that contributed to my publish retry work that stalled in virtual networkStatus issues.