adafruit/Adafruit_Wippersnapper_Arduino #704: Offline mode I2C driver init failure doesn't say which one

tyeth, 2025-02-28:
Using this config, I've only got some of it connected, I'd assumed it would ignore the failed ones and continue, alas it stops and boot loops.
Admittedly I threw some definitions from components repo and the bme280 config at copilot in vscode and it looked convincing but messed up the humidity ones.

```
02:50:50:290 -> [SD] Checksum OK!
02:50:50:290 -> Parsing components array...
02:50:50:291 -> Parsing exportedFromDevice object...
02:50:50:291 -> status LED brightness: 0
02:50:50:291 -> Configuring RTC...
02:50:50:291 -> RTC Type: DS3231
02:50:50:291 -> Begin DS3231 init
02:50:50:292 -> Parsing components array...
02:50:50:292 -> [SD] I2C component found, decoding JSON to PB...
02:50:50:293 -> [SD] I2C component found, decoding JSON to PB...
02:50:50:293 -> [SD] ERROR: Found unspecified SensorType - humidity
02:50:50:294 -> [SD] I2C component found, decoding JSON to PB...
02:50:50:294 -> [SD] ERROR: Found unspecified SensorType - humidity
02:50:50:300 -> [Offline] Attempting to configure hardware...
02:50:50:300 -> [SD] Created new log file on SD card: log_1740622352.log
02:50:50:301 -> -> I2C Device Add/Replace Message Type
02:50:50:301 -> [i2c] Decoding I2cDeviceAddOrReplace message...
02:50:50:302 -> Creating a new I2C driver
02:50:50:302 -> ERROR [HANG]: [i2c] Driver failed to initialize!
02:50:50:302 ->         Did you set the correct value for i2cDeviceName?
02:50:50:302 ->         Did you set the correct value fori2cDeviceAddress?
```

<details><summary>Config Details</summary>
<p>

```json
{
    "exportedFromDevice": {
        "referenceVoltage": 3.3,
        "totalGPIOPins": 18,
        "totalAnalogPins": 4,
        "sd_cs_pin": 23,
        "statusLEDBrightness": 0.5,
        "rtc": "DS3231"
    },
    "components": [
        {
            "name": "BME280 Sensor",
            "componentAPI": "i2c",
            "i2cDeviceName": "bme280",
            "period": 2,
            "i2cDeviceAddress": "0x77",
            "i2cDeviceSensorTypes": [
                {"type": "relative-humidity"},
                {"type": "ambient-temp"},
                {"type": "ambient-temp-fahrenheit"},
                {"type": "pressure"},
                {"type": "altitude"}
            ]
        },
        {
            "name": "SCD40/SCD41 Sensor",
            "componentAPI": "i2c",
            "i2cDeviceName": "scd40",
            "period": 5,
            "i2cDeviceAddress": "0x62",
            "i2cDeviceSensorTypes": [
                {"type": "ambient-temp"},
                {"type": "ambient-temp-fahrenheit"},
                {"type": "humidity"},
                {"type": "co2"}
            ]
        },
        {
            "name": "SEN54 Sensor",
            "componentAPI": "i2c",
            "i2cDeviceName": "sen54",
            "period": 2,
            "i2cDeviceAddress": "0x69",
            "i2cDeviceSensorTypes": [
                {"type": "ambient-temp"},
                {"type": "ambient-temp-fahrenheit"},
                {"type": "humidity"},
                {"type": "pm10-std"},
                {"type": "pm25-std"},
                {"type": "pm100-std"},
                {"type": "voc-index"}
            ]
        }
    ]
}

```

</p>
</details>

tyeth, 2025-02-28:
A healthy config gives this version of the same driver init failed message, but not which one:
```
02:57:56:763 -> Configuring RTC...
02:57:56:763 -> RTC Type: DS3231
02:57:56:763 -> Begin DS3231 init
02:57:56:764 -> Parsing components array...
02:57:56:764 -> [SD] I2C component found, decoding JSON to PB...
02:57:56:766 -> [SD] I2C component found, decoding JSON to PB...
02:57:56:766 -> [SD] I2C component found, decoding JSON to PB...
02:57:56:773 -> [Offline] Attempting to configure hardware...
02:57:56:773 -> [SD] Created new log file on SD card: log_1740622779.log
02:57:56:774 -> -> I2C Device Add/Replace Message Type
02:57:56:774 -> [i2c] Decoding I2cDeviceAddOrReplace message...
02:57:56:775 -> Creating a new I2C driver
02:57:56:775 -> ERROR [HANG]: [i2c] Driver failed to initialize!
02:57:56:775 ->         Did you set the correct value for i2cDeviceName?
02:57:56:775 ->         Did you set the correct value fori2cDeviceAddress?
---- Closed serial port COM32 due to disconnection from the machine ----
```

brentru, 2025-04-14:
Resolved in https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/pull/728

closed 2025-04-14 by brentru