adafruit/Adafruit_Wippersnapper_Arduino #713: Offline timestamp for Soft RTC is not as expected

tyeth, 2025-03-14:
Appears the soft rtc timestamp value is [reading row number]. Ideally it would match the time elapsed

<details><summary>Log with added line breaks showing the 15second break between BME680 reads</summary>
<p>

```
---- Reopened serial port COM46 ----
03:42:01:923 -> [SD] Parsing config.json...
03:42:01:923 -> [SD] Successfully deserialized JSON config file!
03:42:01:924 -> [SD] Checksum mismatch, file has been modified from its original state!
03:42:01:924 -> [SD] Checksum OK!
03:42:01:924 -> Parsing components array...
03:42:01:924 -> Parsing exportedFromDevice object...
03:42:01:924 -> status LED brightness: 0
03:42:01:924 -> Configuring RTC...
03:42:01:924 -> RTC Type: SOFT
03:42:01:924 -> Parsing components array...
03:42:01:924 -> [SD] I2C component found, decoding JSON to PB...
03:42:01:930 -> [Offline] Attempting to configure hardware...
03:42:01:930 -> [SD] Created new log file on SD card: log_6944.log
03:42:01:931 -> -> I2C Device Add/Replace Message Type
03:42:01:931 -> [i2c] Decoding I2cDeviceAddOrReplace message...
03:42:01:942 -> Creating a new I2C driver
03:42:02:709 -> [i2c] Driver initialized and added to controller: 
03:42:02:709 -> bme680
03:42:02:709 -> [Offline] Hardware configured, skipping network setup and running app...
03:42:05:806 -> {"i2c_address":"0x77","timestamp":0,"value":31.42691,"si_unit":"C"}
03:42:05:814 -> {"i2c_address":"0x77","timestamp":1,"value":88.70938,"si_unit":"F"}
03:42:05:822 -> {"i2c_address":"0x77","timestamp":2,"value":36.75614,"si_unit":"%"}
03:42:05:833 -> {"i2c_address":"0x77","timestamp":3,"value":100443,"si_unit":"hPa"}
03:42:05:841 -> {"i2c_address":"0x77","timestamp":4,"value":73.52505,"si_unit":"m"}
03:42:05:849 -> {"i2c_address":"0x77","timestamp":5,"value":48599,"si_unit":"Ω"}

03:42:23:442 -> {"i2c_address":"0x77","timestamp":6,"value":31.28837,"si_unit":"C"}
03:42:23:450 -> {"i2c_address":"0x77","timestamp":7,"value":88.02691,"si_unit":"F"}
03:42:23:458 -> {"i2c_address":"0x77","timestamp":8,"value":36.46287,"si_unit":"%"}
03:42:23:469 -> {"i2c_address":"0x77","timestamp":9,"value":100442,"si_unit":"hPa"}
03:42:23:478 -> {"i2c_address":"0x77","timestamp":10,"value":73.77673,"si_unit":"m"}
03:42:23:486 -> {"i2c_address":"0x77","timestamp":11,"value":52963,"si_unit":"Ω"}

03:42:41:078 -> {"i2c_address":"0x77","timestamp":12,"value":30.82078,"si_unit":"C"}
03:42:41:086 -> {"i2c_address":"0x77","timestamp":13,"value":87.24744,"si_unit":"F"}
03:42:41:097 -> {"i2c_address":"0x77","timestamp":14,"value":36.80429,"si_unit":"%"}
03:42:41:119 -> {"i2c_address":"0x77","timestamp":15,"value":100440,"si_unit":"hPa"}
03:42:41:119 -> {"i2c_address":"0x77","timestamp":16,"value":73.94385,"si_unit":"m"}
03:42:41:123 -> {"i2c_address":"0x77","timestamp":17,"value":56653,"si_unit":"Ω"}

03:42:58:715 -> {"i2c_address":"0x77","timestamp":18,"value":30.48285,"si_unit":"C"}
03:42:58:723 -> {"i2c_address":"0x77","timestamp":19,"value":86.65856,"si_unit":"F"}
03:42:58:731 -> {"i2c_address":"0x77","timestamp":20,"value":37.25008,"si_unit":"%"}
03:42:58:742 -> {"i2c_address":"0x77","timestamp":21,"value":100440,"si_unit":"hPa"}
03:42:58:751 -> {"i2c_address":"0x77","timestamp":22,"value":73.94385,"si_unit":"m"}
03:42:58:760 -> {"i2c_address":"0x77","timestamp":23,"value":59133,"si_unit":"Ω"}
```

</p>
</details> 

On metro RP2350 with BME680

[config.json](https://github.com/user-attachments/files/19240898/config.json)
[secrets.json](https://github.com/user-attachments/files/19240897/secrets.json)
[wipper_boot_out.txt](https://github.com/user-attachments/files/19240896/wipper_boot_out.txt)

Also seen some weird filesystem layout stuff, I've not flash nuked this board, but think it was from when I had a missing SD pin definition from secrets

[image]

tyeth, 2025-03-14:
config attached is correct but somehow the secrets that had the cs pin is corrupted. Been careful to not be powered when inserting sd etc.

Also specifying a period for mux feels weird https://github.com/user-attachments/files/18788402/muxpca-mcp.json [from PR #693 ]

Can't get mux to work, 8port version with all address jumpers soldered, scanned as 0x77, mux initialises but drivers on port 0 don't.
Using standard i2c bus so no pins specified for SDA / SCL
```
{
    "exportedFromDevice": {
        "referenceVoltage": 3.3,
        "totalGPIOPins": 32,
        "totalAnalogPins": 6,
        "sd_cs_pin": 39,
        "statusLEDBrightness": 0.1
    },
    "components": [
        {
            "name": "PCA9548 8-Channel Multiplexer",
            "componentAPI": "i2c",
            "i2cDeviceName": "pca9548",
            "period": 30,
            "i2cMuxAddress": "0x77"
        },
        {
            "name": "mcp9808",
            "componentAPI": "i2c",
            "i2cDeviceName": "mcp9808",
            "period": 10,
            "i2cDeviceAddress": "0x18",
            "i2cMuxAddress": "0x77",
            "i2cMuxChannel": "0",
            "i2cDeviceSensorTypes": [
                {"type": "ambient-temp-fahrenheit"},
                {"type": "ambient-temp"}
            ]
        }
    ]
}
```

brentru, 2025-03-14:
Could you please split the MUX comment off into another issue?

brentru, 2025-03-14:
> Appears the soft rtc timestamp value is [reading row number]. Ideally it would match the time elapsed

This is the expected behavior according to the specifications I implemented for the software rtc. We _could_ add a toggle allowing it to switch between incrementing and elapsed but that would be an enhancement.

tyeth, 2025-03-14:
Yeah sorry, will split this out, was just collecting in one place as same device/session