adafruit/Adafruit_Wippersnapper_Arduino #729: File system instability when attempting to use config.json with Adafruit Feather ESP32-S2 w/BME280 + Adalogger Featherwing

tyeth, 2025-04-11:
This needs shoring up, but this is roughly the outline of what I did...

Recreation steps (windows 11):

1. Board was initially an online wippersnapper device.
2. Used circuitpython.org/downloads via "Install Bootloader" to install a recent bootloader version on feather, using included erase option (not separate erase step).
3. Install offline firmware (v1 or offline-auto-config assets).
4. replace secrets with non-default data
5. Generate config for BME280 and adalogger featherwing on Feather ESP32-S2
6. Copy config.json to device (or write through editor then eject drive).
7. Reboot device and monitor serial port
8. See PCF8523 fail to initialise if v1 offline asset, otherwise if auto-config see no serial output followed by eventual reboot and no config.json file remaining - then attempts to load online mode).

brentru, 2025-04-11:
This should be resolved in https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/pull/728. Please test this as well.

brentru, 2025-04-14:
Resolved by https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/pull/728

closed 2025-04-14 by brentru