adafruit/Adafruit_Wippersnapper_Arduino #736: Offline mode requires users to set secrets values (which are not used) before auto-config kicks in.

tyeth, 2025-04-28:
We need the offline mode to not ask you to update the secrets file before auto config kicks in, it’s not clear to a user that it needs to happen, and is a legacy of Online mode.

brentru, 2025-05-02:
Here, we create the filesystem, potentially including the default `secrets.json` and `config.json` files:
https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/blob/556477721e787c468ce3add7cf44d34fffe1cbca/src/provisioning/tinyusb/Wippersnapper_FS.cpp#L141

If we write to the `secrets.json` file for the first time, we'll raise the error you are seeing here:
https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/blob/556477721e787c468ce3add7cf44d34fffe1cbca/src/provisioning/tinyusb/Wippersnapper_FS.cpp#L159

Whats difficult about modifying this, at the ctor stage, is we aren't yet certain if we're in online or offline mode.

So, a solution may be to remove the check here:
https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/blob/556477721e787c468ce3add7cf44d34fffe1cbca/src/provisioning/tinyusb/Wippersnapper_FS.cpp#L149

and instead rely on the check for default values within the `ParseFileSecrets()` function. That function is called within `WipperSnapper_V2.cpp`, _after_ the configuration file is parsed (and control is returned from the `provision()` to `connect()`
https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/blob/556477721e787c468ce3add7cf44d34fffe1cbca/src/provisioning/tinyusb/Wippersnapper_FS.cpp#L605

brentru, 2025-05-02:
Addressed in https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/pull/744

tyeth, 2025-05-13:
Should be closed by #744 in #748

closed 2025-05-13 by tyeth