adafruit/Adafruit_Wippersnapper_Arduino #747: offline mode PicoW has SD_CS_PIN defined that conflicts with auto_detection of SD

tyeth, 2025-05-12:
PicoW + Adalogger featherwing

The define for SD_CS_PIN (fallback pin) is not correct for our needs. It returns 33, but we need it to be 10.

[image]

As the Adalogger companion board is there we may say that's expected (i.e. a normal pico user may use the default SD card pins and not an adalogger).

The user can fix it with configuring the pin10 / companion board, but it would be nice to attempt to autoconfigure.

Maybe the routine can try pin 10 as a final fallback for RP2 boards, or we use the ws_boards file to give certain wippersnapper expected defaults (i.e. with expected companion boards attached).

To be fair all the arch's could benefit from a list of known sd cs pins that we also try if not in config/use and the default SD_CS_PIN fails (with no config entry).

tyeth, 2025-05-12:
I was using the picow debug target, and I saw output on serial saying the sd failed to initialise. It did take a really long time to finish printing the spi messages on debug serial before finally showing the sd failure on main serial port.
Not yet sure why I didn't see that the first time in non-debug build (from CI)

It's weird brent, as soon as I took a look after dinner at picow in debug build (pio), it spammed a shedload of spi transfer messages over debug serial (pin1/2) as SD was attempting init, then said SD not right over normal serial. It took a while, but I don't know why I never saw those SD messages with non debug CI build of same code, nor the non-picow pio build. Only the slow connection of my serial might have explained it.

I've seen where the drive not reattaching comes from so there's that (if failed SD cs pin then haltErrorV2 doesn't reattach MSC), presumably you're testing mostly with the adalogger rp2040 board instead of pico+adalogger featherwing (which have different SD_CS_PIN definitions so pico shows the issue). See this line for where it happens [haltErrorV2("SD initialization failed.\nDo not reformat the card!\nIs the "](https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/blob/91c2ee8032390b5bb59f16e6c425c9fb6ce7ee97/src/Wippersnapper_V2.cpp#L99)
and this commit for what I did: https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/commit/d9fc3f231ff80633ab3a719ccad055ff6074a43c

brentru, 2025-05-12:
> PicoW + Adalogger featherwing The define for SD_CS_PIN (fallback pin) is not correct for our needs. It returns 33, but we need it to be 10.

I agree with this, do you want to change the `SD_CS_PIN` within the boards file to 10?

tyeth, 2025-05-13:
Problem partially attributed to Between Chair And Keyboard. Ignore previous comment about it ignoring SD_CS_PIN.
The configurator was also set to pin 9 instead of 17 for PiCowbell Adalogger, now fixed, and the boards file has 17 set for Pico 1/2/W

closed 2025-05-13 by tyeth