adafruit/Adafruit_Wippersnapper_Arduino #768: bug: v103-104 Quad Alphanumeric display displays an extra top bar after message "No." and "No. "

tyeth, 2025-07-09:
Quad Alphanumeric display displays an extra top bar after message "No." (bar is position 3) and "No. " (position 4), using left alignment.

[image]

tyeth, 2025-07-09:
there might be something funky with the input too, it has no validation for alphanumeric components, so if you accidentally add a newline you then get a backtick (not the same issue but related - establish then pass to Loren)

tyeth, 2025-07-09:
Also spaces are weird, try sending 4 spaces then a period, then try 3spaces and a period.

tyeth, 2025-07-24:
Seeing the reboot on init messages out of sync too. Was the XIAO, but I confused some LED thing with it actually rebooting.
Has SSD1306 128x64 and a Character LCD 16x2 attached.

<details><summary>See it reboot after init twice then boot okay thrid time, no changes:</summary>
<p>

```
01:21:17:921 -> ERROR [WDT RESET IN -4]: 
01:21:17:922 -> ERROR: Unable to connect to WiFi, rebooting soon...
01:21:18:922 -> ERROR [WDT RESET IN -5]: 
01:21:18:923 -> ERROR: Unable to connect to WiFi, rebooting soon...
01:21:19:922 -> ERROR [WDT RESET IN -6]: 
---- Closed serial port COM52 due to disconnection from the machine ----
---- Reopened serial port COM52 ----
01:21:21:766 -> Found single wifi network in secrets.json
01:21:21:769 -> Adafruit.io WipperSnapper
01:21:21:769 -> -------Device Information-------
01:21:21:769 -> Firmware Version: 1.0.0-beta.108
01:21:21:769 -> Board ID: xiao-esp32s3
01:21:21:769 -> Adafruit.io User: tyeth
01:21:21:769 -> WiFi Network: free4all
01:21:21:769 -> MAC Address: 74:4D:BD:81:B8:A0
01:21:21:769 -> -------------------------------
01:21:21:769 -> ESP32 CPU0 RESET REASON: NO_MEAN
01:21:21:769 -> ESP32 CPU1 RESET REASON: POWERON_RESET
01:21:21:769 -> Generating device's MQTT topics...
01:21:21:769 -> Running Network FSM...
01:21:21:769 -> Establishing network connection...
01:21:21:769 -> Performing a WiFi scan for SSID...WipperSnapper found these WiFi networks:
01:21:24:908 -> free4all (6C:63:9C:23:A3:AB) -60dB (ch6)
01:21:24:908 -> free4all (28:F5:D1:EF:10:12) -68dB (ch1)
01:21:24:908 -> free4all (10:7C:61:82:50:F1) -76dB (ch6)
01:21:24:908 -> Ginger Moggie (20:35:43:4:83:6E) -88dB (ch1)
01:21:24:909 -> Shaun (48:D3:43:A2:44:81) -91dB (ch6)
01:21:24:909 -> VM1451252 (40:D:10:C1:4E:C9) -92dB (ch11)
01:21:24:910 -> TP-Link_BD8E (B0:A7:B9:98:BD:8E) -94dB (ch9)
01:21:25:510 -> Connecting to WiFi (attempt #0)
01:21:31:129 -> Connected to WiFi!
01:21:31:130 -> Connecting to AIO MQTT (attempt #0)
01:21:31:133 -> WiFi Status: 20
01:21:33:919 -> Registering hardware with WipperSnapper...
01:21:33:919 -> Registering hardware with IO...
01:21:33:919 -> Encoding registration request...Encoding registration msg...Published!
01:21:34:022 -> Polling for registration message response...2
01:21:34:624 -> GOT Registration Response Message:
01:21:34:625 -> Hardware Response Msg:
01:21:34:625 ->         GPIO Pins: 14
01:21:34:625 ->         Analog Pins: 11
01:21:34:625 ->         Reference voltage: 2.60v
01:21:34:737 -> Completed registration process, configuration next!
01:21:34:738 -> Polling for message containing hardware configuration...
01:21:34:750 -> cbSignalTopic: New Msg on Signal Topic
01:21:34:750 -> 16 bytes.
01:21:34:750 -> decodeSignalMsg
01:21:34:750 -> cbSignalMsg
01:21:34:750 -> Sub-messages found: 1
01:21:34:750 -> Signal Msg Tag: Pin Configuration
01:21:34:750 -> cbDecodePinConfigMsg
01:21:34:750 -> Configuring digital input pin on D0with internal pull-up enabled
01:21:34:751 -> Interval (ms):0
01:21:34:751 -> Initial Pin Configuration Complete!
01:21:34:751 -> Publishing to pin config complete...
01:21:34:754 -> * NEW MESSAGE [Topic: Signal-I2C]: 
01:21:34:754 -> 87 bytes.
01:21:34:754 -> cbDecodeSignalRequestI2C
01:21:34:754 -> I2C Device LIST Init Request Found!
01:21:34:754 -> EXEC: cbDecodeI2CDeviceInitRequestList
01:21:34:755 -> EXEC: New I2C Port 
01:21:34:755 ->         Port #: 0
01:21:34:755 ->         SDA Pin: 5
[... 134 lines cut ...]
01:22:18:469 -> Hardware configured successfully!
01:22:20:029 -> Registration and configuration complete!
01:22:20:029 -> Running application...
01:22:20:031 -> Sending MQTT PING: * NEW MESSAGE [Topic: Signal-I2C]: 
01:22:20:034 -> 68 bytes.
01:22:20:034 -> cbDecodeSignalRequestI2C
01:22:20:034 -> [app] I2C Device Output Write
01:22:21:127 -> [app] I2C Device Output Write Done
01:22:21:129 -> * NEW MESSAGE [Topic: Signal-I2C]: 
01:22:21:130 -> 56 bytes.
01:22:21:130 -> cbDecodeSignalRequestI2C
01:22:21:130 -> [app] I2C Device Output Write
01:22:21:261 -> [app] I2C Device Output Write Done
01:22:21:261 -> SUCCESS!
01:22:21:261 -> WiFi RSSI: -66
01:22:21:282 -> Executing state-based event on D0
01:22:21:283 -> Encoding pinEvent...Encoded!
01:22:21:283 -> Publishing pinEvent...Published!
01:22:25:775 -> Sending MQTT PING: SUCCESS!
01:22:25:938 -> WiFi RSSI: -68
---- Closed the serial port COM52 ----
```

</p>
</details>

brentru, 2025-08-19:
What is a full-stop?