adafruit/Adafruit_Wippersnapper_Arduino #785: Allow board specific define, for boards without unique ARDUINO_BOARDNAME or no matching entry in boards.txt

tyeth, 2025-07-25:
Some boards use the generic module FQBN or the ARDUINO_BOARD define for a generic module/devkit, when they have differing pinouts. 

We need to be able to distinguish between them during compilation, which we do via ci-arduino build targets with FQBN.
It's possible to override the boards.txt entries for a single board, or even add new boards, using a `boards.local.txt` file installed into the platform folder (where boards.txt resides).

This issue was raised after a user added the DFRobot ESP32C3 which uses the same board define as the Espressif C3 devkit.

I've added an argument to build-platforms.py in ci-arduino, `--boards-local-txt`, with an optional file argument (or defaults to CWD\boards.local.txt). 
- If the argument/flag is supplied, then the **test-examples** function also looks for an example specific boards.local.txt file, or preferably a board specific version `.{PLATFORM}.boards.local.txt` like the .skip files (and .generate which is for UF2)

ci-arduino PR
https://github.com/adafruit/ci-arduino/pull/217

wipper PR
https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/pull/784

I tested with a boards.local.txt at project root, and the updated CI task to use the flag, using a wippersnapper DFrobot C3 specific define (WS_UNIQUE_BOARD_ID) and the plain ESP32C3_DEV separated into two entries in Wippersnapper_Boards.h, one with a broken board ID, and watched them come online with the different id, then swapped the broken one.
Then restored to both being dfrobot c3, but now we can add the c3 devkit or other boards using the ESP32C3_DEV define.

tyeth, 2025-07-25:
Tagging @ladyada as mentioned in the meeting. For now the CI-Arduino PR does not take the change into main branch, but will happily afterwards if it's desired.

ladyada, 2025-07-25:
yep understood!

brentru, 2025-07-28:
@tyeth  Ping me in slack and we'll go over this when you're back from holiday.

tyeth, 2026-10-15:
Closing as fixed by #784

closed 2026-10-15 by tyeth